makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #181: At the end of the file transfer with the MKS WIFI, the card reboot

Dom749s, 2021-03-01:
### Description

At the end of the file transfer with the MKS WIFI, the card reboot 

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Ilya-Pak, 2021-03-01:
#177

Chlorel, 2021-03-03:
Happy You ... You can do file transfert with WIFI ... i cannot !

RAFALFRY, 2021-03-11:
Has someone managed to solve this problem?