makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #182: LVGL-UI and BL_Touch causes lockup at System Boot

RolfZuckowskiUltras, 2021-03-02:
### Description
I did install a 3D-Touch bed leveling sensor on my printer and configured it into the firmware according to the Marlin configuration. The System locks up, if the LVGL-GUI is used in combination with the BL_Touch-Option. The Problem is not existant, if the classic Marlin GUI is used and works fine, or the BL-Touch is deactivated. So it seems to be a bug: Bed leveling vs LVGL-GuI.

The build does always compile fine. If the firmware (LVGL and BL-Touch) gets loaded onto the printer, the system just straight locks up after the update or restart of the printer. The Screen stays just black. Marlin seems not to start up at all. There is no Boot-Message via USB-Serial-Terminal nor the hotend-fan spins up for a short time (which is is normally the case at marlin startup).

### Steps to Reproduce
This Marlin Setup has been used: [See the playground-branch in my repo](https://github.com/RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI/tree/ExperimentalPlayground_DualZ_BLTOUCH)

#### Additional Information
*Printer Setup*
- Two Trees Sapphire Plus with dual Z-Endstopps
- E3D Hemera Hotend
- MKS Robin Nano v1.2 with TFT35
- 3D-Touch Senesor (BL-Touch Clone by Triangle Labs)
- Configuration is verified to work

*Compilation Environment*
- VS-Code with PlattformIO

martinbogo, 2021-03-17:
I'm seeing the same issue.  I am also configuring a printer with a BL-Touch + LVGL-GUI in a similar configuration.

Hardware: MKS Robin Nano v1.2 w/ TFT35 touchscreen
Marlin: Github latest main/master branch as of 1 day ago

Liriel-666, 2021-03-20:
mmh could you reflash the board?

RolfZuckowskiUltras, 2021-03-20:
Reflash? Do you mean a loading the newest firmware completly? Thats what I did.

martinbogo, 2021-03-28:
I have also reflashed/reloaded the board after updating my git to the latest revision.  I still get hard lockups at system boot, and sometimes just after running the BL-Touch level process.   

@MKS-Sean:  **Your patches are EXTREMELY MESSY and are pulling more code than is relevant to a given patch.**  If you look at the latest BL-Touch patches, there are a lot more lines of code pulled in than are part of a given issue.   As an example, look at this latest set, it not only is fixing the image asset issue -- it also changes values in powerloss.cpp AND it messes with pins_MKS_ROBIN_PRO_V2.h and peripheral_pins.c!!!

Please don't use "git commit -a" ( or if you do, make sure that only the code changes relevant to the commit make it in, please) but rather check each file that you're changing into the commit, or things will get very chaotic and hard to fold back into Marlin 2.0 when we present patches to @thinkyhead

e.g.) [this patch here : Add bmp_init_state.bin to the assets array](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/d426c976fa7b2e9aa7fdd78a41b93822a957e03d
)

RolfZuckowskiUltras, 2021-03-28:
> I have also reflashed/reloaded the board after updating my git to the latest revision. I still get hard lockups at system boot, and sometimes just after running the BL-Touch level process.

**_Warning: Other issue, but possibly related to commet above:_**
@martinbogo I did not made up the issue yet and probably not here anyway. But just for your personal notice: If the LVGL-GUI is not used, the Printer is able to start just fine (with BL-Touch configured). It's able to do the leveling process totally fine, but locks up, when the printer is first extruding after starting a print. You're getting a _READ:error:SD read error_ fired out onto the serial console until the watchdog kicks in. I tried messing around at the SD-Speed-Settings and Buffer size a while ago and it was still existant to me. Today I addtionally did activate the SD-Read-Only Mode in conf_adv.h and the error seems to be gone. I did that in a direct marlin fork which I originally made up to counter-check the issue. So if this is a bug, this is a directly Marlin related. 

Didn't made up an _"marlin"_-issue yet, since I'm not quite shure, if that's been the fix and do not have the time to counter check yet. Maybe thats a further hint to you. So if that's a general issue, the board is very likely having problems storing the Bed-Leveling configuration on the SD-Card.

martinbogo, 2021-03-29:
no I have no problems with SD card reads here under lvgl or Marlin TFT

this lock up specifically happens only when BL touch is added to the
configuration, lvgl is used, and bl touch has not yet performed its first
leveling

martinbogo, 2021-03-29:
@RolfZuckowskiUltras : Hi, yes .. I'm aware that it works fine with  TFT_COLOR_UI enabled -- but that's not the point of reporting this bug.  

The bug is that it **does not work** when combined with TFT_LVGL_UI -- and that means we need to fix it.

Liriel-666, 2021-03-29:
Strange is that it worked on my old github build from 2.2.2021 without problems the bltouch and the lvgl-ui

TimCarrDP, 2021-10-08:
Still an issue. Anyone have any idea for a fix?