makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #183: Russian language in the tft_Language_ru.h file

gensoloone, 2021-03-03:
Hello! Tell me please. When editing Russian words (changing capital letters: "инструмент" = "Инструмент", "домой" = "Домой") in the tft_Language_ru.h file, the printer screen displays the changed word without the letter that was changed ("инструмент" = "нструмент", "домой" = "омой"). How do I fix it so that it would be displayed correctly on the screen?

Здравствуйте! Подскажите пожалуйста. При редактировании русских слов (изменение заглавных букв: "инструмент" = "Инструмент", "домой" = "Домой") в файле tft_Language_ru.h на экране принтера отображается измененное слово без буквы которую меняли ("инструмент" = "нструмент", "домой" = "омой"). Как исправить так, что бы  правильно отображалось на экране?

MKS Robin Nano v1.2 and MKS TFT35

hakimio, 2021-03-03:
Are you using "UTF-8" encoding when saving?

gensoloone, 2021-03-03:
> Are you using "UTF-8" encoding when saving?

Yes. In VS Code, I have UTF-8 read and write by default. I will try to edit the file today via Notepad++, Maybe VS Code somehow saves the file incorrectly.

Да. В VS Code у меня по умолчанию стоит чтение и запись UTF-8. Попробую Сегодня отредактировать файл через Notepad++, Может VS Code как-то неправильно сохраняет файл.

Malderin, 2021-03-22:
Заглавные русские буквы почему то не поддеживаются, по этому, когда я переводил все это для добавления в прошивку, использовал только строчные буквы.

JPatison, 2021-09-20:
> Заглавные русские буквы почему то не поддеживаются, по этому, когда я переводил все это для добавления в прошивку, использовал только строчные буквы.

Что значит "почему-то не поддерживаются" - давайте разбираться почему. Я думаю в файле шрифта нет нужных глифов, нужно пересобрать файл шрифта. В каком он формате непонятно, я пробовал lv_font_conv и заливать в прошивку, после этого дисплей вообще не отображает ни русские ни английские символы. Пробовал китайскую утилиту LvglFontTool 0.4 она делает тоже самое. Кроме файла шрифта BIN создается C файл, куда его лепить неизвестно.

closed 2023-02-16 by gensoloone