makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #184: Printing doesn't start using SD in lvgl ui

DvoraNoob, 2021-03-03:
every time I put something to print it gives it home and for printing and it only happens in lvgl ui

Chlorel, 2021-03-03:
hi, have you done a PID auto tune once ? Some time print does not start because the board have pain to get the right temps ...
I use MKS Nano 1.2 with TFT 35 screen on my printer ... I use to LVGL for my UI .. and i just had the print start problem once when i had bad PID values for my hotend

DvoraNoob, 2021-03-03:
I did PID auto tune aready

DvoraNoob, 2021-03-27:
Uploading 20210327_194345_1_1.mp4…

DvoraNoob, 2021-03-27:
https://user-images.githubusercontent.com/62312359/112737069-7cd0d980-8f36-11eb-959a-fdb0c50d9d49.mp4

ColdBeer72, 2021-04-02:
I've got the same problem. Everything works, even writing printing commands through pronterface, but when selecting a file to print, the printer warms bed and hotend, and time counter stops when reached temps but it can't start to print.

ColdBeer72, 2021-04-02:
My actual configuration.h:
```
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW       // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  #define FIL_RUNOUT_PIN MT_DET_1_PIN
  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M25"
```
I found that:

1. It doesn't matter if I define or not define FILAMENT_RUNOUT_SENSOR
2. It doesn't matter if I choose FIL_RUNOUT_STATE HIGH or LOW
3. The only things makes the printer works is to physically unplug the filament sensor from the board.

Why the hell?

ColdBeer72, 2021-04-02:
PROBLEM SOLVED!!! I needed to change **/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h**:
```
  //#define MT_DET_2_PIN                      PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define MT_DET_PIN_INVERTING             true  // LVGL UI filament RUNOUT PIN STATE
```
I don't know why the hell this change (taken from [issue#66](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/66)). First I tried with just the second line (as I want to use MT_DET2_PIN for my 3dtouch sensor), but it still doesn't work until you comment the first line.

closed 2021-05-05 by DvoraNoob