makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #186: Mks wifi gcode output feg. eeprom setting via wifi not usb

Darkraven2k3, 2021-03-09:
Hi,

I got a robin nano v2 with ts35 monitor and mkswifi
 
Now thanks to Luca I got esp3d fw installed and so fare working with webui.

The problem is, that most of the Gcode I send via wifiterminal only report back:OK.

When I have my printer connected via USB and use slicer3d to see terminal than I can input my commands in webui and my printout will be on my pc.

Ia there a way to get something like dual output aka. Return information back over wifi serial and also USB?

MKS-Sean, 2021-03-20:
You mean to return the command and ok sent by the web page？ like this:
send: M105
receive: ok

Darkraven2k3, 2021-03-20:
Yes exactly. Only host connected via USB is receiving the data output.

Darkraven2k3, 2021-03-20:
Good would be full compatibiöity for mks wifi v2 aka esp8266. And also enable wifi for marlinui.