makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #189: How do I flip the display 180 degrees?

Kn9z, 2021-03-14:
This is it: (#define TFT_SCALED_DOGLCD 1
#if TFT_SCALED_DOGLCD
   #define GRAPHICAL_TFT_ROTATE_180)
Re-rotates the display in x-only directions.
How to flip the y-axis?

Kn9z, 2021-03-14:
#ifndef TFT_ROTATION
  #define TFT_ROTATION TFT_ROTATE_180
#endif
does not work

MKS-Sean, 2021-03-20:
@Kn9z Are you using the latest version? Which ui is used? You only need to configure in configuration.h:
#define TFT_ROTATION TFT_ROTATE_180

Kn9z, 2021-03-20:
latest version

Kn9z, 2021-03-20:
> @Kn9z Are you using the latest version? Which ui is used? You only need to configure in configuration.h:
> #define TFT_ROTATION TFT_ROTATE_180

MKS-Sean, 2021-03-20:
Can you take a photo for me? After flipping the display photo, which motherboard are you using?

Kn9z, 2021-03-20:
@MKS-Sean Robin nano v1.2