makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #193: Error compiling after activating 3d Touch ...

themacboy, 2021-03-16:
Hello !

I just configured my 3d touch in my Mks Robin Nano v.3 board (to be used on Z min, following mks wiki) but I receive next error while compiling:

[image]

I tryed to compile using Marlin Color UI, and all was perfect (the compilation is ok) and the 3d touch works perfect.

Is the any problem in the MKS UI core? Some one with same board and 3d touch in zmin has the same problem?

Thanks in advance !

themacboy, 2021-03-16:
you can reproduce de error using you example files for Ender 3 with bltouch and tmc2209

themacboy, 2021-03-17:
Error comes due a missing param on file "draw_level_settings.cpp" line 69:

[image]

themacboy, 2021-03-17:
Looks like there is needed a boolean value, not sure if true or false

Chlorel, 2021-03-17:
the missing param is fallse

Chlorel, 2021-03-17:
or you can delete the line ... iniit not need to be forced by manuall launch snce it can be done iin interface

Chlorel, 2021-03-19:
look's like this issue is solved :) feel free to close the issue :)

closed 2021-04-13 by MKS-Sean