makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #195: Problem with merge request #190 ...

themacboy, 2021-03-17:
Good monring,

after you merge the #190 FAN 1 and FAN 2 stop working. (acording PCB):

[image]

I have reset this change and FAN 2 works again.

About FAN 1, Im not sure how to make it run.

Im using Robin Nano v3 and Marlin Color UI.

themacboy, 2021-03-17:
Not sure if it is also related to #191

ProjectPatatoe, 2021-03-17:
@themacboy Can you turn turn on `FAN_SOFT_PWM`. This setting can be uncommented in Configuration.h around line 2500. I don't have a second board to test with.

I am also using Robin Nano v3 and Color UI.

ProjectPatatoe, 2021-03-17:
Also, Were both fans working and correctly assigned before? If it did, it would seem that your board doesn't match either of the three versions of the schematic. Were you using a precompiled marlin?

Liriel-666, 2021-03-17:
mmh on my robin nano v3 with github and MKS UI from 17.3 the fan works but only when i set ist to 128 and above.
with 127 and lower it doesnt move

ProjectPatatoe, 2021-03-18:
Yea I had the same problem with only on or off. I measured with an oscilloscope and found that the processor wasn't even putting out a PWM signal. This fixed for me once I turned on `FAN_SOFT_PWM`.

Before #190 Fan1(UI) controlled Fan2(physical) and Fan2(UI) controlled Fan1(physical)

themacboy, 2021-03-18:
I just set:
 
#define FAN_PIN                             PC14   // FAN0
#define FAN1_PIN                            PB1  // FAN1

and fans are not working.

Then I set:

#define FAN_SOFT_PWM

and fan still not working.

Then negative, that solution dint work for me. :(

ProjectPatatoe, 2021-03-18:
under what combinations are working for people?
The full working combination for me is 
```
#define FAN_PIN PC14 // FAN0
#define FAN1_PIN PB1 // FAN1
#define FAN_SOFT_PWM
```
I am using platformio to compile

themacboy, 2021-03-18:
> 
> 
> under what combinations are working for people?
> The full working combination for me is
> 
> ```
> #define FAN_PIN PC14 // FAN0
> #define FAN1_PIN PB1 // FAN1
> #define FAN_SOFT_PWM
> ```
> 
> I am using platformio to compile

for me only work fan1 (fan2 acording pcb)

#define FAN_PIN PB1  // FAN0
#define FAN1_PIN PC14 // FAN1

In all other sets my fans din't work. Nothing

ProjectPatatoe, 2021-03-19:
FINALLY I found the reason! The pins have actually been changed from revision 3.0.002 and 3.0.003. It seems that pins have actually changed as far I as can see. It probably should've been a 3.1 actually since it requires a different pins file. 
[image]

Liriel-666, 2021-03-19:
ah thats explain why i had with the 02.02.21 github build Fan 1 on the Fan 2 Port.
Now it correct and soft pwm works vor me until the Board goes black/dead with yesterday build

MKS-Sean, 2021-03-20:
@Liriel-666 Is there any problem with the current version of the fan?

Liriel-666, 2021-03-20:
> 
> 
> @Liriel-666 Is there any problem with the current version of the fan?

mmh that i mut enable soft pwn was in the github version from 18.3.
but i can not test newer firmware because of that
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/197

ProjectPatatoe, 2021-03-20:
@Liriel-666 Which revision of the board are you using? Is it revision 001 or 002? It is on the back on the bottom right corner.

ProjectPatatoe, 2021-03-20:
Because of the mismatched documentation on the hardware repo, I can't really tell if its revision 002 or 003 that changed the fan pin. https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/issues/14

Liriel-666, 2021-03-20:
> 
> 
> @Liriel-666 Which revision of the board are you using? Is it revision 001 or 002? It is on the back on the bottom right corner.

its a 002 but with that https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/197
its useless