makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #196: MKS Nano V3 and MKS Wifi Module

FabioMC, 2021-03-18:
I have just added to my MKS Nano V3 a MKS Wifi module. According the information available I changed the UI from COLOR_UI to LVGL_UI since the module only work in this setup. My panel is a MKS TS35.

When compiling the new firmware to include the Wifi communication I receive the error message shown in the attached picture. Any hint on how to fix this error?

[image]

MKS-Sean, 2021-03-20:
Sorry, now i have fixed it

FabioMC, 2021-03-20:
@MKS-Sean . After your last update the compilation is presenting lots of warning message. See below into the picture. The compilation finishes but for some reason the board is not updating. The TS35 stays with 100% done. The makerbase logo do not appears and so do not start updating assets.  I upload the older firmware using TFT_COLOR_UI and everything works which means that the board is ok. It is happening just when I use TFT_LVGL_UI. I appreciate if you could revise and see what is happening.

[image]

FabioMC, 2021-03-20:
@MKS-Sean. You can deconsider my last comment. Apparenlty when I changed the pin definitions to enable the PWC I also enabled the MKS Test var. I supposed that this was the problem, because since I disabled it everythin worked well.

closed 2021-04-13 by MKS-Sean