makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #197: Black screen after autolevel.

anon-anonim, 2021-03-18:
Hi!
 The following problem occured
After I start the autoleveler (induction sensor), the alignment starts (a message is displayed, the alignment is in progress), when the script finishes, the screen turns black and does not respond to commands.
[https://drive.google.com/file/d/1vPBj23IiMTdTv4RuX1571QN34ZmsZThH/view?usp=sharing](url)

Liriel-666, 2021-03-18:
i have the same problem.
Board start an make bltouch test but no screen or speaker.

i make leveling too then the image of the buttons was defekt then nothing.

Liriel-666, 2021-03-18:
i think something ist with the bootloader or Firmware.
even updatign firmware doesnt work because the file is not renamed

MKS-Sean, 2021-03-20:
@anon-anonim When did you use the firmware submitted, can the latest submission solve your problem?

Liriel-666, 2021-03-20:
when he has the same problem what i have  with black screen and no reaktion 

> 
> 
> @anon-anonim When did you use the firmware submitted, can the latest submission solve your problem?
when he has the same problem what i have  with black screen and no reaktion  then he can not update any firmware because the system doesnt start anymore and with that no firmware update will install.

black screen after autolevel that result in no start of the board. only the 3,3V led is on but nothing happens.
I try  to install a new firmware but nothing happen.
it doesnt install it and it is to see that the firmware file is not renamed

ProjectPatatoe, 2021-03-21:
Are you both using the `TFT_LVGL_UI`, or the color/classic UI?

Liriel-666, 2021-03-21:
tft_lvgl_ui

> 
> 
> Are you both using the `TFT_LVGL_UI`, or the color/classic UI?

i use TFT_LVGL_UI

ProjectPatatoe, 2021-03-21:
If you want to just get the printer to work right now, I would suggest trying with `TFT_COLOR_UI` for now. I currently have BLTouch working with the default Marlin interface.
I believe the LVGL UI doesn't fully support auto leveling quite yet. At the very least, I switched away from it since I needed advanced pause and manual leveling for a while.

Liriel-666, 2021-03-21:
> If you want to just get the printer to work right now, I would suggest trying with `TFT_COLOR_UI` for now. I currently have BLTouch working with the default Marlin interface.
> I believe the LVGL UI [...]

I think you dont read all postings!
It doesnt Install any Firmware anymore!
No rename of the Firmware file that show that the firmware is installed. 
it doenst show anything on the Display.

Nothing worked without the BL Touch that is in endless diagnostic mode because it seams the Board reboot endless.

I tried 3DAYS!!!!!! to get a net firmware installed over all usb und mircro sd ports!
nothing! usb b doesnt work because even windows doesnt found the port.

It is impossible to install a new firmware!!

How should i install a Firmware update when Micro SD, USB A and USB B not working?
can you answer this?

and as a hint! the with the Github from 2.2 was it possible to use auto leveling without any problem with LVGL UI.

steve007iii, 2021-03-24:
I have the same problem.

anon-anonim, 2021-03-27:
> @anon-anonim When did you use the firmware submitted, can the latest submission solve your problem?
I installed a new version today, the error remained. The auto-calibration field disappears, the screen glows black.

Liriel-666, 2021-03-27:
> > @anon-anonim When did you use the firmware submitted, can the latest submission solve your problem?
> > I installed a new version today, the error remained. The auto-calibration field disappears, the screen glows black.

How you get a firmware update install ed?

anon-anonim, 2021-03-28:
My English is very bad. Therefore, I recorded a video.
https://drive.google.com/file/d/1u9mQXmjjdrhivtHjgtq0QtogIl9wTuAE/view?usp=sharing

Liriel-666, 2021-03-28:
> 
> 
> My English is very bad. Therefore, I recorded a video.
> https://drive.google.com/file/d/1u9mQXmjjdrhivtHjgtq0QtogIl9wTuAE/view?usp=sharing

ah its different to my board.
i can get start it because it hang in a bootloop

Luppi1324, 2022-07-06:
Hi
I have the same problem.
I still have Marlin 2.0.8 on my Robin_Nano V1.2

Fixes the problem with the latest Marlin version
Thanks

Bkara1981, 2023-03-26:
Same problem, power is on, but no display and no light or sound from lcd, cannot update firmware via sd card because nothing is happening. I think something happened to bootloader.

quocphan16033011, 2024-05-21:
> > If you want to just get the printer to work right now, I would suggest trying with `TFT_COLOR_UI` for now. I currently have BLTouch working with the default Marlin interface.
> > I believe the LVGL UI doesn't fully support auto leveling quite yet. At the very least, I switched away from it since I needed advanced pause and manual leveling for a while.
> 
> hahaha i dont get on that idea with install a new firmware! BECAUSE I had it 3 Days ago!
> 
> But I think you dont read all postings! It doesnt Install any Firmware anymore! No rename of the Firmware file that show that the firmware is installed. it doenst show anything on the Display.
> 
> Nothing worked without the BL Touch that is in endless diagnostic mode because it seams the Board reboot endless '(a boot Loop)! because the diagnostic mode ist only short when the touch gets power.
> 
> I tried 3DAYS!!!!!! to get a new firmware installed over all usb und mircro sd ports! nothing! usb b doesnt work because even windows doesnt found the port or show a hint of it.
> 
> It is impossible to install a new firmware!!
> 
> How should i install a Firmware update when Micro SD, USB A and USB B not working? can you answer this?
> 
> and as a hint! with the Github from 2.2 was it possible to use auto leveling without any problem with LVGL UI.
Did you have any way to fix it after 3 years ? . I have just gotten this problem.