makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #198: Icon modification and font modification

soo246, 2021-03-19:
I started the Marlin firmware for the first time, but I couldn't answer it even though I looked for it, so I ask.

1.To MKS-Robin-Nano-Marlin2.0
Can't I delete the icon itself??
2. Can I change the icon back to another layout??
3. I want to edit the text, but is there any way??
ex) Home →Homeing
4. Is there any way to remove the text part if it is impossible to edit the text?

Here is where I downloaded the firmware.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

The program is
visual Studio code

If you know how, please let me know in detail.

Chlorel, 2021-03-19:
if you compile a firmware from the folder, you have to put the icons and fonts in an "assets" folder ... 
To modify the positioning in the interface, you need to know how to program a little bit, and modify the interface which is in the /lcd/extui/lib/mks_ui folder to set the functions called when you select a given icon.

soo246, 2021-03-20:
그럼 글자를 수정하거나 없애는 방법을 알려주실수있나요??
ex) move → printing
이라는 글자를 바꾸고싶어서요

soo246, 2021-03-22:
Thanks to
I made the settings.
Thank you.

closed 2021-04-13 by MKS-Sean