makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #200: MKS UI - Tool Filament - Two Extruders - Single Nozzle

FabioMC, 2021-03-19:
Tool filament is not working for the second extruder with a single nozzle. When you choose the second extruder and requires the load action the heating information is always zero althoug the hotend is heating up. Since the value do not increase the load action do not perform.
[image]

Darkraven2k3, 2021-03-20:
Also have the same problem. I define temp sensor with same pin for heaterhotend in pins.h 

Also opened and issues for mixing mode with 2 steppers where only 1 extruder is selectable.

FabioMC, 2021-03-20:
Any feedback from @makerbase-mks  or @MKS-Sean ?

Darkraven2k3, 2021-03-20:
Till now they might look after it. I got the problem that I can't cobtroll second extruder via filamentload or extrudermenu. Only one is working for me. 

I also suggested to implement mixing extruder correct via lvlgui so that it might be possible to change mixing while printing like it is on marlinui

Darkraven2k3, 2021-03-20:
All I want is to use correct mixing hotend like e3d cyclops or geeetech 2 un 1 out hotend.

closed 2021-04-23 by MKS-Sean