makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #203: H43 menu on ts35 possible

Darkraven2k3, 2021-03-20:
I realy like the menu design from new h43 screen for creality. 

Is it somehow possible to get this cool menu on ts35 (landscape)‽

Statusscreen would be realy nice on startscreen. Cause now on mksui there are only 3 entry's with lots of free designspace

Chlorel, 2021-03-20:
I think that with knowledge and skill anything is possible, but to be able to do what you want to do, you may need to know what the menus look like 

And not everyone keeps the 3 buttons on the home screen of the MKS interface, there are people who add useful information :)

Darkraven2k3, 2021-03-20:
I saw u have also your own fork. Also realy like the new h43 interface.

U think it will be possible to adopt ui elements from this, or do u have maybe a modded ui u wanna share or can Pont a link for me?

I got realy no knowledge un lvlgui.

Chlorel, 2021-03-20:
in my fork i have two particular branches ... one for my friend's Ender 3 and one for my Wanhao D12 ... feel free to have a look at them ... every thing done here is open source, so all i do and upload to my github is free to use also :)