makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #205: MKS Robin Nano V1.2 i MKS Wifi Module

RAFALFRY, 2021-03-20:
Hi,
I am having trouble printing via wifi.
The printing process hangs.
[image]
Configuration files in the attachment:
[Marlin.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6176468/Marlin.zip)

Thanks

Chlorel, 2021-03-21:
look's like you have a extruder temperature problem too, bed goes to 50° but extruder is setted to 0°

RAFALFRY, 2021-03-21:
The bed heats up quickly.  After that, everything freezes.  The only problem is when printing via wifi.

MKS-Sean, 2021-04-13:
@RAFALFRY Hello, this looks like an error in the file data transmitted by wifi, you can try to format the sd card, or change to a more stable network

RAFALFRY, 2021-04-14:
@MKS-Sean Hello, The card format doesn't help.  When the nozzles and the bed are heated earlier, the printout is done.
[image]