makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #207: Options: MIXING_EXTRUDER reason of compilation crashing

madurani, 2021-03-21:
Hello All,
I am tried compile Mks-Robin-Nano-Marlin2.0-Firmware-master, because I installed on my printer second extruder with nozzle 2IN1 and I want test mix-color printing and this functionality isn't according MKS-support supported in Mks_Nano35_V2.x firmware.

I only configured main parts and it worked till I enabled options:
#define MIXING_EXTRUDER
after enabling this it did not work anymore. Compilator reported issue:

Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:1057:6: error: #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."
     #error "For MIXING_EXTRUDER set MIXING_STEPPERS > 1 instead of EXTRUDERS > 1."

Configured was only:
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define SERIAL_PORT 3
#define EXTRUDERS 2
#define SINGLENOZZLE
#define MIXING_EXTRUDER
#define MKS_ROBIN_TFT35
#define TFT_LVGL_UI
#define TOUCH_SCREEN

When is options #define MIXING_EXTRUDER commented, then compilation finish success.

Darkraven2k3, 2021-03-21:
The problem is u can only set Single nozzle with 2 extruder or 1 extruder and then mixing mode.

But here I also opened an issue cause I can't control second extruder in mksui. In marlinui mixingmode is working.

madurani, 2021-03-22:
yes one extruder + mix mode >>> compilation work. But I am not sure, how is needed in this case setup cura and octoprint. 
cura > 2 extruded + shared nozzle? , octoprind > 2 extruder and shared nozzle? will be it work?

Could you please somebody paste here configuration of mks bltouch for mks-robin-nano 1.2v, because YouTube video from mks isn't exactly for mentioned board and it is complicate for me, because I am beginner with marlin configuration/compilation.
I have setup bltouch as ZMAX and no as Z safe homing.

Chlorel, 2021-03-22:
@madurani have a look on my fork, in branch Wanhao D12/230 ... I setted it up for run with 3D-Touch or BL-Touch

madurani, 2021-03-22:
**@Chlorel** please provide link, because I can't find it.

Chlorel, 2021-03-23:
@madurani here : https://github.com/Chlorel/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Wanhao-D12/230

madurani, 2021-03-23:
**@Chlorel**
What worst can be happens, when I incorrectly configure marlin and install firmware to robin-nano(with marlin I am beginner)? Isn't possible, that I will lost possibility again re-install new firmware and board stay hang(black screen/dead)?

Chlorel, 2021-03-23:
@madurani i dont think you can destroy your board by just setting wrong marlin, but i am not sure about this, i never setted it wrong for my firmwares

madurani, 2021-03-23:
**@Chlorel**
In case, that I will not able correctly configure  marlin firmware, can I switch back to mks robin nano v.2 firmware via sdcard?

Chlorel, 2021-03-23:
if you have MKS Robin nano v2 board with TS35, you can switch back to any MKS Robin nano v2 firmware 
But i have no MKS Robin nano V2 .. Just a V1.4 and a V3 and a E3P

madurani, 2021-03-23:
**@Chlorel**

I can't find in Configuration.h file part for definition of bltouch array  
// Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION  20
  #define RIGHT_PROBE_BED_POSITION 190
  #define FRONT_PROBE_BED_POSITION 20
  #define BACK_PROBE_BED_POSITION  200

madurani, 2021-03-23:
**@Chlorel**

I installed  Mks-Robin-Nano-Marlin2.0-Firmware to MKS robin nano 1.2v. At first glance everything looks fine, but auto leveling not work properly. I check all your example and all are with bltouch installed as z-min-endstop. I have z-min-endstop active and bltouch is plug to z-max-endstop. Can you help me with this kind of configuration?
In mks firmware I have set up:
 Z_SAFE_HOMING                  0       # 0:Disable; 1:Enable
USE_ZMAX_PLUG  1                       # 1:used; 0:noused
Z_MIN_PROBE_ENDSTOP_INVERTING 0
Z_MIN_PROBE_PIN_MODE           2       # 0 : NULL; 1: ZMIN; 2: ZMAX

With this configuration of bltouch work properly in mks firmware, but all configuration which I find for marlin, are with bltouch plug in z-min and serve a z-min-endstop/ or z-save-homing.

Chlorel, 2021-03-24:
for my BL-Touch configuration i use : 
in configuration.h
#define USE_PROBE_FOR_Z_HOMING
#define Z_PROBE_PIN PC4 
#define BLTOUCH
#define MULTIPLE_PROBING 2
#define PREHEAT_BEFORE_PROBING
#define AUTO_BED_LEVELING_BILINEAR
#define GRID_MAX_POINTS_X 5

in configuration_adv.h
#define BABYSTEPPING

madurani, 2021-06-13:
I reopen this issue, because I have again problem with MIXING_EXTRUDER and compilation crashing. I download new source code from https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware.git. But compilation crashed when I enable option DISTINCT_E_FACTORS(I need it because I have two different extrudes). In older version of MKS-marlin is compilation success with same configuration.

I have enable these options:
```
 #define EXTRUDERS 1
//#define SINGLENOZZLE
#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2       
  #define MIXING_VIRTUAL_TOOLS 16  
  #define DIRECT_MIXING_IN_G1    
  #define GRADIENT_MIX          
  #if ENABLED(GRADIENT_MIX)
    #define GRADIENT_VTOOL       
  #endif
#endif
#define DISTINCT_E_FACTORS

```
Error outputs:
```
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:28:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:3305:4: error: #error "MIXING_EXTRUDER can't be used with DISTINCT_E_FACTORS. But you may use SINGLENOZZLE with DISTINCT_E_FACTORS."
 3305 |   #error "MIXING_EXTRUDER can't be used with DISTINCT_E_FACTORS. But you may use SINGLENOZZLE with DISTINCT_E_FACTORS."

```

closed 2021-09-06 by madurani