makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #208: Add auto tune PID

Dtcreation, 2021-03-22:
### Description

Auto tune PID

### Steps to Reproduce

1.Add a menu for auto tune PID

**Expected behavior:** Do not use the host to do PID

**Actual behavior:** No possibility

#### Additional Information

Chlorel, 2021-03-22:
You can use Pronterface, but connected with usb ... and set your PID with a command like : M303 E<n> S<t> C<c> 
with : 
n : extruder number ( -1 for bed ... 0 for first ... 1 for second ...)
t : your optimal temperature ( i set it to 200 )
c : autotune test count ( i use 4 ... but you can set more )

Dtcreation, 2021-03-22:
Yes, but il will be nice to have it dirrectly on the screen

Chlorel, 2021-03-22:
i tried ... but ... not that easy to do ... every string is sended to LCD screen with a send serial command ...