makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #211: After homing z-axis not work

madurani, 2021-03-24:
I compiled Mks-Robin-Nano-Marlin2.0-Firmware and installed it to MKS-Robin-Nano-1.2v. 
After starting printer all axis x,y,z work correctly on both direction +/-, but after "homing-all", nozzle move to homing position and after this operation, z-axis dont move any more. Next uncrect thing is, that x,y are on position 0,0, but on z-axis is nozzle approximately 2mm over bed.

When I switch back with MKS-firmware to board, then all work correctly

Chlorel, 2021-03-24:
try to uncomment USE_PROBE_FOR_Z_HOMING

madurani, 2021-03-24:
**@Chlorel**
I enabled it, but finis with error. Z_SAFE_HOMING is needed and I haven't bltouch use as endstop:
#error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."

madurani, 2021-03-24:
Bingo, all work, I am no sure what exactly helped me, but work without recabling of bltouch:

- auto leveling
- mixing material(virtual extruded)
- and communication of firmware with octoprint

Later I will paste here options which I enabled, but I have elese next questions.

Darkraven2k3, 2021-03-25:
Yes. Please give hints. I also looking for mixing mode cause of 2 in 1 nozzle. U have marlinui or mksui?

Hope changes will also work for robin nano v2

madurani, 2021-03-25:
Till now I used mks firmware v2 on mks-robin-nano-v1.2, but here wasn't possible use mixing mode of color. Code:
M163 S0 P0.7
M163 S1 P0.3
M164 S2
wasn't supported. From these reason I am trying switch to  Mks-Robin-Nano-Marlin2.0-Firmware
Problem wasn't version of board, but kind of cabling/installation of bltouch in my case. I have z-min-endstop in z-min port and bltouch in z-max and z-homing wasn't enabled in in my case.
Usually is bltouch in port z-min, z-homing is enabled and z-min-endstop isn't active.

My changes in Configuration.h.(I am absolutely beginner in Marlin and take options from different youtube videos)
#define EXTRUDERS 1
//#define SINGLENOZZLE

#define MIXING_EXTRUDER
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  #define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
  #define GRADIENT_MIX           // Support for gradient mixing with M166 and LCD
 

#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN PC4

#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET { -23, 0, -2.71 }
#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR false
#define INVERT_E0_DIR false
#define INVERT_E1_DIR true

//#define MIN_SOFTWARE_ENDSTOP_Z

#define AUTO_BED_LEVELING_BILINEAR

#define Z_SAFE_HOMING

closed 2021-03-25 by madurani