makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #212: [Feature Request] MKS LVGL UI - Better UI for Longer Gcode Filenames

mfedoseev, 2021-03-24:
### Description

The MKS LVGL UI looks nice, is themable, but the file list is almost unusable. None of my filenames fit. The two vertical lines of text under each icon fit only 12-14 characters in total, not including the ".G" at the end.

### Steps to Reproduce
Using MKS Robin Nano 2.0 / TS35.

1. Create (or rename) one or more .gcode files.
2. Name them with 15+ characters. I.e. "storage-box-full-height-v2.gcode".
3.  Put on an SD card and insert into the printer.

**Expected behavior:** [What you expect to happen]

Expecting fully readable file names - longer strings or at least scrolling.

I think the SD file list needs to change from square tiles to a full-width vertical list, where the file names are longer, single lines of text. Similar to regular Marlin UI. And the ".g" does not add any info to the user, so it needs to go.

**Actual behavior:** [What actually happens]

Seeing a grid of things named like this:
"storage-
box-fu~.g".

#### Additional Information

I already reflashed my board to Marlin UI, but you can see for example, here, how names are VERY short. It's almost like MS-DOS days:
https://3dtoday.ru/upload/posts/main/QON0IHIyzbwA0VabBB2r6aPUClDTQsytsTqs091IxmHNqcqmMD.png
https://user-images.githubusercontent.com/52313714/103137046-9adc3300-46d6-11eb-8943-c05f214694b8.jpg

MKS-Sean, 2021-04-13:
Hello, I will try to make it into a rolling form

WashingtonJunior, 2021-06-05:
By the way, how to support long filenames when sending files through the wifi module? Files gets an old 8.3 filenames sizes when sent that way.

Thanks!