makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #213: BL-Touch Probe - Missing data after Autoleveling

FabioMC, 2021-03-24:
When I use the Autolevel Tool menu, as soon as the leveling process finishes, the Z-Offset screen appears but there a failing information related to the nozzle temperature I supposed. I am sending a picture with the missing data. Just for information my printer has a singlenozzle and two extruders configuration.

[image]

madurani, 2021-03-24:
I have same issue with same printer(nozzle 2IN1, two extruders)

Chlorel, 2021-03-24:
you havent added "bmp_init_state.bin" to your asset folder before flashing the firmware

Chlorel, 2021-03-24:
For Z-Offset setting menu, only one extruder temp is show ... not both

FabioMC, 2021-03-25:
No. The file is in the folder as you can see in the image attached. After compilation I copy the assets folder and *.bin file to flash the board. Never had to inlcude manually a file to assets folder.

About the temperature a single nozzle has only one temperature anyway.
[image]

Chlorel, 2021-03-25:
@FabioMC but the  place where you show me 'no data' is the place where bmp_init_state.bin is ...
[image]

[image]

madurani, 2021-03-25:
I can confirm that, mentioned file is include in folder aset on sd-card and data missing on screen too:
[image]

madurani, 2021-03-25:
What mean this?
[image]
because there was same file changed before 7 hours ago

madurani, 2021-03-25:
I tried download https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/archive/refs/tags/v2.0.5.3.zip
but compilation finish with error:
platformio.ini:
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
**default_envs = mks_robin_nano35**

madurani@pc-marek /virtualbox/Marlin/Mks-Robin-Nano-Marlin2.0-Firmware-2.0.5.3 $ p
latformio run -e mks_robin_nano ; echo "done" >|/tmp/ipc
**Error: Unknown environment names 'mks_robin_nano'.** Valid names are 'SAMD51_grandcentral_m4, STM32F103RC_btt_USB, sanguino644p, rumba32_f446ve, STM32F407VE_black, STM32F103RC_btt, BIGTREE_BTT002, STM32F103VE_longer, DUE_debug, rambo, ARMED, teensy35, STM32F103RE_btt, esp32, mks_robin, STM32F103CB_malyan, FLYF407ZG, STM32F103RE_btt_USB, STM32F103RC_fysetc, linux_native, FYSETC_S6, rumba32_mks, at90usb1286_dfu, sanguino1284p, DUE_USB, STM32F401VE_STEVAL, melzi_optiboot, mks_robin_lite, STM32F103VE_GTM32, FYSETC_F6_14, FYSETC_F6_13, STM32F103RC, DUE, BIGTREE_GTR_V1_0, STM32F103RE, chitu_f103, melzi, STM32F103RC_btt_512K, STM32F4, at90usb1286_cdc, mks_robin_pro35, BIGTREE_SKR_PRO, LPC1769, mks_robin_mini, teensy31, mega2560, include_tree, STM32F7, mks_robin_lite3, STM32F103RC_btt_512K_USB, LPC1768, mega1280, **mks_robin_nano35**, jgaurora_a5s_a1'

madurani, 2021-03-25:
I downloaded new source code via git clone, copy there Configuration.h file, compile it and problem with missing image was solved.

FabioMC, 2021-03-25:
The same for me. After downloading the new source code everything is working fine.

Ryszard-wpl, 2021-03-26:
I had the same issue but aditionaly I encountered many resets (crush) of mainboard during seting up z-offset. After autolevel or direct from config -> BL touch menu. I need to make it fast to store z-offset. Many times I was making leveling with paper and have seen black screen. I need to make it over again untill success. Is it related to lack of icon - I afraid it is not...

MKS-Sean, 2021-04-13:
@Ryszard-wpl  I will test this problem and solve it

Ryszard-wpl, 2021-04-15:
crash problem is solved with last update with lack of icon. There is only request to put manual 5 point leveling after autolevel with BL Touch. Reducing level differences make prints beter and faster I think. Is it call UBL ? When you will issue this function to TFT UI? Thanks in advance.