makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #214: Not realy a Issue, but help is needed

Chlorel, 2021-03-25:
I've been trying to modify the manual levelling screen for a few days now to add an automatic levelling icon, and possibly add an icon for the Z-Offset adjustment as well, this would allow to keep both ways of adjusting the bed...

But
When I want to use the current code for the automatic leveling called from the leveling menu, the printer does the 25 test points ... Then instead of putting the Z offset setting screen, I get a black screen ... and a few moments later a reboot ... 
I took the instructions to launch the test of the tray and put them directly in my manual setup window, which makes me lose the work of @MKS-Sean on the waiting interface and the call of the Z-Offset setup...
For the call of the Z-Offset, I tried by positioning my call in the window of manual leveling, i get directly a black screen + reboot, I also tried directly since the menu TOOLS ... there it works ... let's say 'almost' .... on the other hand it works very well from the setup menu where the window is originally coded ...
I don't have any debugging tools, so I tried undressing the function ... 
The call of the init procedure happens normally ...
It's as soon as I try to draw the first button that the reboot happens ... 

Does anyone have any idea what is going on? 
and why a screen is displayed from one place and not at all from another?

thank's for any hint

MKS-Sean, 2021-03-26:
Hello, if you can, share your warehouse link, I can modify it with you, it is best to have a small video to show the problem you encountered

Chlorel, 2021-03-26:
As always, all the changes I make to the code are on my github: https://github.com/Chlorel/Mks-Robin-Nano-Marlin2.0-Firmware/tree/Wanhao-D12/230 
Not the master branch ( this one is to have a clean copy of your's ... Got to the WanHao-D12/230 branch

By default, the configuration.h file is for the version with the BL/Touch on Z_MAX for the Wanhao D12 230

The goal is to gain a place on the TOOL menu to add the icon of a neopixel setting screen 

There is an issue in the latest version on Malderin's modification ... when you are in print menu and you click on 'Z' .... It rebooted ... I commented the case in draw_printing.cpp

Chlorel, 2021-03-26:
Oh yes, if you mount the firmware on one of the cards (Nano 1.4) tell me what you think of my ready_print screen :)

Chlorel, 2021-03-28:
@MKS-Sean , can you please confirm if i can plug a ST LINK v2 to J11 on MKS Robin Nano 1.2 ? And if it is a way to plug a JTAG ?
Thank's

MKS-Sean, 2021-04-13:
@Chlorel Hello, sorry for being late, there is a reserved SWD interface on the nano motherboard, is this what you need?

Chlorel, 2021-04-13:
@MKS-Sean to be late is not a problem :) Think you have more to do as just play on old robin nano 1.2 board :)
Yes, SWD is supported by ST-Link v2 so i think debuging session will be easyer :)

MKS-Sean, 2021-04-13:
I used J-Link's SWD interface to debug it

Chlorel, 2021-04-13:
i seen that in platformIO.ini, but a bit late (i pursached the st-link v2 day's before)