makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #216: MKS Robin Nano V3.0 + H43 fw 1.30 + WiFi + MKS TS35 V2.0 +Serial Communication + MKS servo 57B V2.0

xNeurosynapticx, 2021-03-26:
[Robin_nano_v3.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6213757/Robin_nano_v3.zip)

https://drive.google.com/file/d/1-xtct92XxQM_H869CIb3FjITY8hm4ljc/view?usp=sharing

FULLY functional firmware for Robin nano v3.0 + H43 fw 1.30 + TS32 + WIFI +Serial comunication

willngton, 2021-03-26:
That is great, thanks for sharing, got mine and start to get ready for the upgrade.

Did you get any luck with marlin bug-fix?

No problem with MKS version, got some issue with Marlin, will try and compare later

Thanks again .

Darkraven2k3, 2021-03-27:
Hi is it possible that u might try to build with robin nano v2 Board? Can u provide source?

U know if it is possible to rotate screen for ts35 to landscape or is h43 only up down

spammchik, 2021-04-01:
Is it possible for such a configuration to work?
MKS Robin Nano V3.0 + H43 fw 1.30 + Serial Communication + USB support
(no WiFi, no MKS TS35 V2.0)
Unfortunately, I could not enable USB support,
#define USB_FLASH_DRIVE_SUPPORT - doesn't work. The firmware could not compile, gave an error:
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp: In function 'void tft_lvgl_init()':
Marlin\src\lcd\extui\lib\mks_ui\tft_lvgl_configuration.cpp:134:15: error: 'USB_POWER_CONTROL_PIN' was not declared in this scope
134 | OUT_WRITE(USB_POWER_CONTROL_PIN, HIGH);
And when I did this:
[platformio]
default_envs = mks_robin_nano_v3_usb_flash_drive_msc
the firmware could not compile, gave an error.
At the moment, I have never seen USB support (for printing from a flash drive) + Display H43 working on this board.

xNeurosynapticx, 2021-04-01:
it is possible to have usb support (mks nano v3.0)
here is a firmware for an Ender 3 that works (no H43).

[Robin_nano_v3 (usb).zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6244582/Robin_nano_v3.usb.zip)

At the moment I am working on a custom user interface, my main goal is to build a complete MKS system for a cnc machine, when finished I will work on 3d printers

[image]

xNeurosynapticx, 2021-04-06:
> Is it possible for such a configuration to work?
> MKS Robin Nano V3.0 + H43 fw 1.30 + Serial Communication + USB support
> (no WiFi, no MKS TS35 V2.0)
> Unfortunately, I could not enable USB support, [...]

DONE

[Robin_nano_v3.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6266483/Robin_nano_v3.zip)

spammchik, 2021-04-08:
I ran this firmware on Robin Nano V3:

> https://drive.google.com/file/d/1-xtct92XxQM_H869CIb3FjITY8hm4ljc/view?usp=sharing
> FULLY functional firmware for Robin nano v3.0 + H43 fw 1.30 + TS32 + WIFI + Serial comunication

The H43 display is working!
But there is a problem with printing through a computer via RepetierHost.
Often when printing, the printer stops for a few seconds and then continues to move.
However, there is an error in RepetierHost: Communication timeout - reset send buffer block
I tried to do different port speed in firmware (115200 and 250000) and in RepetierHost, but it didn't help.
Can you fix this?

xNeurosynapticx, 2021-04-08:
[image]

[CFFFP_MKS_H43上盖.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6280201/CFFFP_MKS_H43.zip)

I printed this file in repeater with that configuration and It gave me several errors, the serial communication was disconnected multiple times with this error. 

Resend: 991
13:00:55.026 : N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.126 : Resend: N991 G0 X66.347 Y136.313 F2400*124
13:00:55.126 : UpdateScreenVPData Screen: 1 VP: 8192 VP: 8196 VP: 8256 VP: 8260 VP: 9984 VP: 0 UpdateScreenVPData done
13:00:55.126 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.127 : (3) # 130 79 75 # >
13:00:55.127 : (3) # 130 79 75 # >
13:00:55.127 : (3) # 130 79 75 # >
13:00:55.127 : (3) # 130 79 75 # >
13:00:55.127 : (3) # 130 79 75 # >
13:00:55.127 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.127 : Resend: 992
13:00:55.226 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.226 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.226 : Resend: 992
13:00:55.326 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.326 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.326 : Resend: 992
13:00:55.425 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.425 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.425 : Resend: 992
13:00:55.525 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.525 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.525 : Resend: 992
13:00:55.625 : Resend: N992 G1 F1800 X60.793 Y136.313 E640.78567*56
13:00:55.625 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.625 : Resend: 992
**13:00:55.726 : Receiving only error messages. Stopped communication.**
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 991
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 992
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 992
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 992
13:00:55.880 : Error:Line Number is not Last Line Number+1, Last Line: 992

I will take a look at this problem.

I was able to print the entire file with pronterface, which can be a compatibility problem when sending the codes to the board.

spammchik, 2021-04-09:
I have also tried this firmware. And I found that it also had this problem. 
> MKS Robin nanov3.0+H43+WIFI+TS35v2.0+ext Drivers (MKSservo57B)+USB support+Serial Communication(repetier,CNCjs, etc...)
Printer stops when printing from Repeater Host with error "Communication timeout - reset send buffer block"
Also, I did not see any USB drive or SD card files in Windows.
Also, the display language after restarting the printer is always reset to Chinese, English is not saved.

spammchik, 2021-04-14:
It seems that increasing the size of the buffers should help.

spammchik, 2021-04-14:
I see you are well versed in the firmware for Robin Nano V3.
If you know the answer to this question, please answer, because everyone is silent.
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/226

jpeero, 2023-09-25:
where is the download for the firmware?  i am new to this and cant get my robin nano v3 working at all from what makerbase provides as firmware...