makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #218: Warning emitted during compile of current "master" branch

martinbogo, 2021-03-29:
### Description

Compiler throws the following warning when compiling the current "master" branch in PlatformIO on VSCode

Compiling .pio\build\mks_robin_nano35\lib814\lvgl\lv_draw\lv_draw.c.o
.pio\libdeps\mks_robin_nano35\lvgl\src\lv_core\lv_obj.c: In function 'lv_deinit':
.pio\libdeps\mks_robin_nano35\lvgl\src\lv_core\lv_obj.c:130:5: warning: implicit declaration of function 'lv_log_register_print_cb'; did you mean 'lv_obj_get_event_cb'? [-Wimplicit-function-declaration]
     lv_log_register_print_cb(NULL);
     ^~~~~~~~~~~~~~~~~~~~~~~~
     lv_obj_get_event_cb
     
### Steps to Reproduce

Clone current branch, compile using the default configuration.

**Expected behavior:** [What you expect to happen]

This warning should not be thrown... there is something wrong with the usage/inclusion/declaration of lv_log_register_print_cb

Chlorel, 2021-03-29:
hi @martinbogo 
this is a warning because you dont have a function for logging ( what's completely normal on a micro controler )
in .pio/libdeps/mks_robin_nano35/lvgl@src-0cc1a42ddb0ca1a5f25965e4fae93318/src/lv_core/lv_obj.c

at line : 127 modify the lv_deinit(void) like this : 

void lv_deinit(void)
{
    lv_gc_clear_roots();
	#if LV_USE_LOG 
      lv_log_register_print_cb(NULL);
	#endif
    lv_disp_set_default(NULL);
    lv_mem_deinit();
    lv_initialized = false;
    LV_LOG_INFO("lv_deinit done");
}

dokkio-martin, 2021-04-07:
Confirm, this fix works.