makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #221: Robin TFT35 V1.0 screen orientation during the boot

ostapoff, 2021-04-07:
I am running the Robin Nano V1.2 + Robin TFT35 V1.0 screen shipped with the Sapphire Plus 3D printer.
The screen orientation is upside down which is configured by setting the following option:
#define TFT_ROTATION TFT_ROTATE_180

The screen works fine when booted but during the boot I can see the "Booting..." or "TFT Updating..." phrases up side down. On native Sapphire Plus firmware it is corrected with this robin_nano35_cfg.txt option:
>cfg_screen_overturn_180 0xEE 

How can this be configured in Marlin 2.0 with defines?

MKS-Sean, 2021-04-14:
In marlin you only need to define:
#define TFT_ROTATION TFT_ROTATE_180
Then after updating the firmware for the first time, reboot, so that the interface of boot and homepage will be flipped 180 degrees

ostapoff, 2021-04-14:
That's the problem. The interface of the homepage and the booted Marlin is displayed OK but the boot interface is not.

ostapoff, 2021-04-14:
You may see how ti works here:
[https://youtu.be/DY2-9FK_Ue4](https://youtu.be/DY2-9FK_Ue4)
On 3rd second the **Booting…** message appears for a fraction of a second upside down

Stefan-van-Hattum, 2022-06-30:
I am having this issue as well, did you solved it and how ?