makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #222: M117 Show messages when printing from OctoPrint / USB Serial

dokkio-martin, 2021-04-07:
The MKS LVGL EXTUI interface does not have any way to show M117 messages on the main display ( tool/settings/printing ) when the file being printed is coming from the USB port via USB Serial.  ( i.e. Octoprint ) 

I have the nozzle temperature and heated bed temperature showing in the main display.

Is there a way in the MKS LVGL interface to have it either switch to the menu screen that is normally shown during an SD card print, or a way to print M117 messages being sent to the printer on this menu screen?

Chlorel, 2021-04-08:
@dokkio-martin 
That's not that easy to do in the way the LVGL_UI is build, and in the way marlin send's message to screen's :)

dokkio-martin, 2021-04-08:
well I just dug into the code and it seems that it's listening on the
serial port for a message... the m117 message is printed clearly there as
an output from G-Code... exactly the same way the temperature is

mlewis-everley, 2021-11-14:
I just got my Sapphire running via Octoprint and spotted this issue too. Would be great if there was a way to display the status screen when an M117 is issued.