makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #224: Printer reboots random on MKS Robin Nano 1.2 when WIFI module connected

BaltasarParreira, 2021-04-09:
Since I don't know if it is Robin Nano 1.2 firmware issue or a MKS WIFI module I leave here the link to the issue I report on the MKS WIFI git:

https://github.com/makerbase-mks/MKS-WIFI/issues/19

RAFALFRY, 2021-04-09:
This problem has been around for a long time. :(

BaltasarParreira, 2021-04-10:
I see, but on latest Marlin 2.0 Bugfix Branch this doesn't happened however with this I can't print anything since the printer moves like crazy all around on the X,Y and Z but never really prints nothing.
I tried also the disable Watchdog that some users report, but even don't have the reboots I get always a freeze in the end of uploading a file that only power off and on again the printer solves it.

BaltasarParreira, 2021-04-12:
So nobody wants to help solving this?
I started investigating the source code looking for problems and a way of improvement.
I already found some code typos and committed them, but the problems still persists so far.

chshkhr, 2021-04-12:
I have MKS PWC and my Robin Nano 1.0 powers off when I connect to it via WiFi. While on stock firmware, everything works fine

BaltasarParreira, 2021-04-12:
@chshkhr Yes the problems of rebooting and file transfers not reaching the end and so on are only on the merged version with Marlin 2.0.
All the previews versions are working ok, but in my case I need the Marlin version because it's the only way I can use my 2209 drivers in UART mode, previous versions don't allow this as well.

MKS-Sean, 2021-04-14:
@BaltasarParreira I need to know the problematic version you are using, I need to test it, the wifi transmission problem when using robin_nano_v1.2

BaltasarParreira, 2021-04-14:
The version I'm using is the one from the source here, I just clone and compile it.
Do you want the compiled file is that it?

DvoraNoob, 2021-04-15:
Same with Robin Nano V2, but without a wifi and principally in the Autoleveling Menu

BaltasarParreira, 2021-04-16:
Just an update.
Since there was new updates to the source I just did a git pull and compiled and install new firmware to check.
Reboots once in a while still happened just leaving the printer on without doing nothing even not touching the screen.
 
So I decided to do another test and this time I comment the #define MKS_WIFI_MODULE just to see if it the problem was really related with the WIF module. 
And yes, with this define comment out I have zero problems and everything works normal, of course I have none of the WIFI functionalities but that was expected.

chshkhr, 2021-04-23:
When Cura plugin is UnConnected Robin Nano stops rebooting

BaltasarParreira, 2021-04-23:
Hummm, I didn't remember doing that test, I can test here to see if I get the same behavior.
Anyway If we have MKS WIFI module is for using it with Cura so It needs to be connected, otherwise I remove it and back to old annoying way of printing, either put file in SD card, remove SD card from PC, put SD card in printer and so on. 
Or having a longer USB cable to the printer and print thought USB witch is no so good as somethings if PC locks, or crashes, printing bye bye and goes to trash whatever was started.
The amazing good point of using MKS plugin on Cura with the WIFI module is avoiding all this, so this is mandatory that should work and I can't understand why all this time and nobody cares.
Why selling a product that doesn't work !!!

chshkhr, 2021-04-24:
I agree. That was just for information about the issue The product works ok with the old firmware (IAR 1.0.5, 2.0.3.3). I hope the issue will be solved in the new Marlin 2.1.x.x too.

BaltasarParreira, 2021-04-24:
I downgrade until almost initial commit of Marlin 2.0 and all test compiled firmware's none worked 100% with WIFI, can you direct me to a specific source that works and I use to compile for my board/printer?

chshkhr, 2021-04-24:
I use my fork from this https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware.
I fixed a small problem with More UI https://github.com/chshkhr/MKS-Robin-Nano-Firmware

BaltasarParreira, 2021-04-24:
Haaa... but that is the closed Marlin version, I need the new as I'm using UART mode so I need to setup a lot on the configuration files. That version can't do that using the old CFG...TXT file, right?

chshkhr, 2021-04-24:
Yes, you are right. I set Vref manually. You have to choose WiFi or UART. I prefer WiFi

BaltasarParreira, 2021-04-24:
With the normal recent Marlin 2.0 I have UART for the 4 drivers on PA6 without problem, and us the older 2.1.x ESP3D for WIFI, the only problem with this setup is that I can't send files to print directly from Cura, I have to write them on PC and them from the ESP3D interface print them via WIFI.

MKS-Sean, 2021-04-30:
@BaltasarParreira 
Hello, the firmware I compiled with the code here does not show the connection restart problem you mentioned, and I am not sure what the cause is. Maybe it is a problem with the maple library, you can try to use the arduinostm32 library, as long as you define in platformio.ini:
default_envs = mks_robin_nano35_stm32
Recompile and test it

BaltasarParreira, 2021-04-30:
I had the default_envs = mks_robin_nano35_stm32 in some compilation, but with so many tests I did I don't remember anymore if it was with the MKS source.
I'll try it again anyway to make sure and let you know later.

chshkhr, 2021-05-03:
> @BaltasarParreira
> Hello, the firmware I compiled with the code here does not show the connection restart problem you mentioned, and I am not sure what the cause is. Maybe it is a problem with the [...]

I have compiled sources from the current repository with `default_envs=mks_robin_nano35_stm32` but the problem persists: my Robin Nano 1.2 with PWC powers off when trying to connect WiFi from Cura plugin

BaltasarParreira, 2021-05-03:
@MKS-Sean  I compiled with the ENVS you suggested and I still have the same problem.
I can do a movie to demonstrate it later.

DvoraNoob, 2021-05-05:
i bought a wifi module for my Robin Nano V2 and i'm with same problem

ElBarbuo, 2021-05-06:
Hi, i have the same problem with my robin nano 1.2 + wifi module + 2209 and mks last marlin firmware(2.0.8). I can´t send files via wifi and printer reboot randomly, sometimes i can resume print, sometimes nope. I got it to work with that firmware(https://github.com/Sergey1560/Marlin_FB4S) but with color ui(not lvgl) and without wifi transfer function... I think I will not buy MKS electronics in the future... too much work!

epl692, 2021-05-08:
I am also having the issue, Robin Nano 1.2 + wifi module + 2208 drivers. Unit does fine until I try to connect cura to it. Just pulled the latest firmware tonight to try to fix it. Any Ideas?

volvofixthis, 2021-05-10:
Same situation, robin nano v1.2. When cura connected, printer reboots after some time.

dominic0012, 2022-04-09:
I have a Robin Nano v2 board. Same problem, reboots every so often. Think I'll give MKS a miss from now on too