makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #226: Robin Nano V3 firmware via USB drive

spammchik, 2021-04-12:
What needs to be done to be able to flash the Robin Nano V3 using the USB drive?
That is, I have the "Robin_nano_v3.bin" file, copy it to the USB drive, insert the USB drive into the Robin Nano V3.
Turn on the power supply to the electronics.
Next, automatic flashing should begin.
How do I enable this feature?

willngton, 2021-04-12:
I am having the same problem, wont start to flash when power supply removed and connecting back.

The workaround would be reset the board, wither from the TFT re-seat button or the on inside the board.

spammchik, 2021-04-16:
The answer is this: to flash through USB, you must remove the SD card!
Power up the electronics.
Insert USB drive, and then press reset.
The firmware will start!

closed 2021-04-16 by spammchik