makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #227: Unable to compile Firmware for MKS Robin Nano V3.0 with vscode

Redboxon, 2021-04-16:
### Description
I tried to load up the latest available firmware I downloaded here in VS-Code. I came to the point of compiling but then it would just spit out an error saying I should select an STM32F4... board in the "Tools->Boards" menu. I haven´t found such a foulder. Now I´ve tried to completely Delete and redownload the Firmware. Now when I hit compile it does seemingly nothing...
I haven´t tried changing anythig in the firmware (besides the Board from V2 to V3)
 

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

stream2me, 2021-04-20:
Did you change the build environment in VScode to something like "mks_nano_v3..." or the default one in platformio.ini?

Redboxon, 2021-04-20:
I left it on the default. It could also be just me just too dumb. I asked a friend who then compiled for me with no problems. I guess it's just a problem with my VS Code installation or I'm just too dumb

closed 2021-04-20 by Redboxon