makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #228: Compilation Error

FabioMC, 2021-04-16:
@MKS-Sean, tried to compiled the last firmware version you published but I am receiving this error message during the process:

Compiling .pio/build/mks_robin_nano_v3_usb_flash_drive/src/src/lcd/extui/lib/mks_ui/draw_filament_change.cpp.o
Compiling .pio/build/mks_robin_nano_v3_usb_flash_drive/src/src/lcd/extui/lib/mks_ui/draw_filament_settings.cpp.o
Marlin/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp: In function 'void event_handler(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp:60:76: error: expected identifier before '(' token
   60 |       if ((thermalManager.temp_hotend[0].celsius >= EXTRUDE_MINTEMP) && && (queue.length <= (BUFSIZE - 3))) {
      |                                                                            ^
Marlin/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp:60:75: error: expected ')' before '(' token
   60 |       if ((thermalManager.temp_hotend[0].celsius >= EXTRUDE_MINTEMP) && && (queue.length <= (BUFSIZE - 3))) {
      |          ~                                                                ^~
      |                                                                           )
*** [.pio/build/mks_robin_nano_v3_usb_flash_drive/src/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp.o] Error 1
=================================================== [FAILED] Took 380.18 seconds ===================================================

Environment                        Status    Duration
---------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive  FAILED    00:06:20.182
=============================================== 1 failed, 0 succeeded in 00:06:20.182 ===============================================

Can you help?

MXMRR, 2021-04-18:
Hi, I got the same error : 

Marlin\src\lcd\extui\lib\mks_ui\draw_extrusion.cpp: In function 'void event_handler(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\lib\mks_ui\draw_extrusion.cpp:60:76: error: expected identifier before '(' token
       if ((thermalManager.temp_hotend[0].celsius >= EXTRUDE_MINTEMP) && && (queue.length <= (BUFSIZE - 3))) {

I'm compiling for a MKS Nano V1.2 board. The error appear when I set :
#define EXTRUDERS 2
#define SINGLENOZZLE

Thanks for your help

tomjrusso, 2021-04-20:
this error is appearing for all versions V1.2, V2, V3 this error is serious keeping all original file just configuring for two extruders with a single nozzle this error appears or 30% of the printers are not working

MKS-Sean, 2021-04-23:
@FabioMC @MXMRR @tomjrusso Hello, I fixed the error.

closed 2021-04-23 by FabioMC

madurani, 2021-07-08:
> @FabioMC @MXMRR @tomjrusso Hello, I fixed the error.

How did you fix issue?  Because I downloaded source MKS-2.1.3 and after enabling mentioned two parameters:
```
#define EXTRUDERS 2
#define SINGLENOZZLE
```
 on MKS Nano V2 I got same error:

```
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_language.cpp.o
Marlin/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp: In function 'void event_handler(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/draw_extrusion.cpp:60:76: error: expected identifier before '(' token
       if ((thermalManager.temp_hotend[0].celsius >= EXTRUDE_MINTEMP) && && (queue.length <= (BUFSIZE - 3))) {
```