makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #229: Robin Nano V3 printer stops before it starts.

666cin, 2021-04-17:
### Description

Unable to print from usb drive. I have bug fresh MKS Marlin v2.0.x with usb enabled and ups and pwc module + filament sensor on robin nano v3. 
while choosing file from usb drive, printer is heating, leveling and then it stops. I can hit resume but it is not resuming. Before it stops bed is going down and up for ~10mm. 
when I will send the same file trough usb cable and simplify it is printing without and issue. 
Was trying to uncomment power loss but it did no change the behavior.

666cin, 2021-04-17:
Update. After changing TFT_LVGL_UI to Marlin Color UI everything is working fine. I can print via usb w/o any issues.

lilive1372, 2021-04-21:
Hello I have the same problem, but I would like to use the original interface and not marlin ui.
If I connect the signal from the ups to Mt det 2, my printing pauses to replace the filament.
How can we connect the signal of the UPS?

666cin, 2021-04-21:
> Hello I have the same problem, but I would like to use the original interface and not marlin ui.
> If I connect the signal from the ups to Mt det 2, my printing pauses to replace the filament.
> How can we connect the signal of the UPS?

Use Marlin ui. It’s much better. With many more options

lilive1372, 2021-04-22:
Hello

Sorry for my english
I don't like marlin UI and I use MKUI's wifi.
Using the BL touch and the PWC, I plugged in the PWC like this:
#define SUICIDE_PIN PW_OFF
#define KILL_PIN PC4 in robin nano_V3.h and set the UPD #define POWER_LOSS_PIN PW_DET but at power failure the Z does nothing

MKS-Sean, 2021-04-23:
@666cin @lilive1372 You can send your configuration file and I will test it.

lilive1372, 2021-04-23:
Hi, 
Thanks for your help.
I put you the configuration.h, configuration.adv.h and robin nano V3.h files
Configuration for Core XY
[Robin nano V3, BLtouch, TMC2209, MKS PWC and MKC UPS.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6366182/Robin.nano.V3.BLtouch.TMC2209.MKS.PWC.and.MKC.UPS.zip)

lilive1372, 2021-05-09:
Hi, 
Have you been able to try the configuation?