makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #230: Swap out Boot Logo with a custom one

Christanoid, 2021-04-20:
Hello,

first of all my hardware: mks robin nano v3 with the TS35-V2.0.

As far as I have understood all images that are being displayed on the display have to be compiled into a .bin file according to the tutorial found in the git (convert it with the LVGL Converter using color format "True Color" and the file output format "Binary RGB565").

I now have my image, a 480x320 png of my boot logo, ran it through the LVGL Converter with the settings as described, replaced the bmp_logo.bin in the assets folder, copied it to my USB and had the printer update via USB (using the mks_robin_nano_v3_usb_flash_drive_msc environment in PlatformIO).
Yet when the printer boots it still has the default MKS logo.

It would be nice if I could replace the boot logo so what am I doing wrong here?

Thanks for the help.

stream2me, 2021-04-20:
Not sure i understand you right. Did you run the update via USB pendrive or USB serial?
I normally run the update via USB-pendrive.
If i wish to replace the assets i copy the assets folder(with new images, but same names) and the mks_robin_XXX.bin to root of pendrive. Insert pendrive into the bord and power on the printer. During the update you can see something that tells you that the assets are also updating. 
Maybe you have to rename the folder on pendrive from "_assets" to "assets".

Christanoid, 2021-04-20:
Hi,
I run the update via USB pendrive. I am doing the update the same way as you, so put the assets folder (not "_assets") on the flash drive together with the mks_robin_xxx.bin, putting the flash in the printer and powering it on. The update shows Erase Sector 1 - 11 aswell as one loading bar saying Updating xx% ...
I don't see anything saying updating images or the similar.

stream2me, 2021-04-20:
I tried to reproduce your steps and encountered the same problem. But somehow I solved it.
I think the problem was that I used vanilla marlin on the first run.
During my trials, I switched back to mks-marlin. And I am pretty sure that is the difference.
After this the printer reboots, the boot logo appears and now the asset update starts.
I also found that it is enough to just create the assets folder, updating the firmware is not necessary.

mks-viva, 2021-05-19:
You only update assets folder(has replace logo bin file).

closed 2021-08-10 by mks-viva