makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #236: G28 G29 M500 no funciona

JLuisCS, 2021-04-24:
Hello, I'm new at this. I recently acquired the Robin Nano V2.0 and I am a little disappointed. Even so, I trust in the constant support of this plate.
I have several problems but one of them is that, the autoleveling Command Settings> G28, G29, M500 does not level me the bed and returns to the previous screen.
Congratulations, you are doing a great job, continue like this.
Best regards.

themacboy, 2021-04-24:
Working fine for me in all versions.

Missatge de JLuisCS ***@***.***> del dia ds., 24 d’abr. 2021
a les 22:32:

JLuisCS, 2021-04-25:
Can I send you configuration.h so you can give you a review?
Thank you

JLuisCS, 2021-04-26:
ok also confircuration_adv.h.Many thanks

ColdBeer72, 2021-04-27:
Hi! I've got the same problem (more or less the same): If I define in configuration.h a grid of more than 5x5 points, **the board reboots** when finished G29 (**just when it supossedly saves the values**).
There is no error at that point through the screen nor through Pronterface, but that is my problem. Maybe the same as @JLuisCS

JLuisCS, 2021-04-28:
> […](#)

> […](#)

You can please send me the config.h and the config_adv.h to compare with my config.h
Thanks for your time

MKS-Sean, 2021-04-30:
@JLuisCS Hello, you can refer to the configuration here

MKS-Sean, 2021-04-30:
@ColdBeer72 Hello, which motherboard are you using

ColdBeer72, 2021-04-30:
> @ColdBeer72 Hello, which motherboard are you using

Robin Nano V3

I must say I found MY problem: I followed the instructions from your bltouch for Robin Nano v3, and the problem was with line:
`#define Z_PROBE_END SCRIPT "M500\nM501"`
If I comment it, then have no troubles at all (I could manually save it with M500, or with LVGL menus: Tools/Autoleveling/save or save eprom in config menu.
It should be interesting that somebody could test that configuration (Z_PROBE_END_SCRIPT and big mesh) to have more data.

JLuisCS, 2021-04-30:
MKS Robin Nano v2.0 mksts35 v2.0

madurani, 2021-08-29:
I have same problem on robin navo v2 and command M500. Reason is unkorect eeprom chip installed on this equipment. Vendor write, that on board is 4KB eeprom, but I have there only 2KB eeprom installed. For more details  https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/257#event-5199883830. 
**Description of issue:**
Via command M500 should be data stored from G29 to eeprom, but it not work, because marlin need 4KB and reachable are only 2KB. 
Command M501 should be read wrote data from eeprom, but there aren't any data, so all data from G29 are lost.

There are two solutions:

1.enable  #define FLASH_EEPROM_EMULATION and store data there. The command #define Z_PROBE_END SCRIPT "M500\nM501" after G29 work then correctly.
2. Comment //#define Z_PROBE_END SCRIPT "M500\nM501" and start G29 before every printing(store data via display and icone eeprom-store or gcode console not work). Please be aware, that in this case after every turn off board are all data from G29 lost.