makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #237: Trying to complile FW for Sapphire Plus printer - error

Excaliburst, 2021-04-27:
### Description
During compile in PlatformIO I see this specific error:
stm32f1/pins_MKS_ROBIN_NANO.h: No such file or directory

Furthermore I have 49 warnings
{
	"resource": "/c:/Users/skjor/OneDrive/Documents/IT/_3D_Printer/_TwoTrees Sapphire Plus/Firmware/Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen/Marlin/Configuration.h",
	"owner": "C/C++",
	"code": "59",
	"severity": 8,
	"message": "function call is not allowed in a constant expression",
	"source": "C/C++",
	"startLineNumber": 154,
	"startColumn": 5,
	"endLineNumber": 154,
	"endColumn": 12
}

Later another error is thrown
{
	"resource": "/c:/Users/skjor/OneDrive/Documents/IT/_3D_Printer/_TwoTrees Sapphire Plus/Firmware/Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen/Marlin/Configuration.h",
	"owner": "C/C++",
	"code": "29",
	"severity": 8,
	"message": "expected an expression",
	"source": "C/C++",
	"startLineNumber": 550,
	"startColumn": 19,
	"endLineNumber": 550,
	"endColumn": 20
}

### Steps to Reproduce

1. Download Marlin 2.0.7.2. - and unpack it to a folder
2. Install Viaul Studio Code
3. Go to folder where you downloaded Marlin
4. Make changes to Configuration.h and Configuration.adv.h.
5. Try to compile

**Expected behavior:** 
Following this video https://www.youtube.com/watch?v=GbHsNCDrTpY
with exacting directions on how to configure the Sapphire Plus CoreXY printer with Marlin FW, I expected to be able to compile a .bin file.

**Actual behavior:** PlatformIO thows the above errors.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[Configuration_files.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6384966/Configuration_files.zip)

MKS-Sean, 2021-04-30:
Hello, you can use the code under the current warehouse to compile. The configuration is as follows:
platformio.ini:

default_envs = mks_robin_nano35

Configuation.h:
#define SERIAL_PORT 3
#define MKS_ROBIN_TFT35
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define TFT_LVGL_UI
#define TOUCH_SCREEN

Excaliburst, 2021-04-30:
Thanks MKS-Sean,

I have already made your suggested changes. But still get this error during compile.
*****
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34:0,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:525:12: fatal error: stm32f1/pins_MKS_ROBIN_NANO.h: No such file or directory
   #include "stm32f1/pins_MKS_ROBIN_NANO.h"      // STM32F1                                env:mks_robin_nano35
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CalledProcessError: Command 'arm-none-eabi-g++.exe -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DHAVE_SW_SERIAL -DMCU_STM32F103VE -DSS_TIMER=4 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\skjor\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\skjor\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\skjor\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\skjor\OneDrive\Documents\IT\_3D_Printer\_TwoTrees Sapphire Plus\Firmware\Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 286:
    apply_features_config()
  File "C:\Users\skjor\OneDrive\Documents\IT\_3D_Printer\_TwoTrees Sapphire Plus\Firmware\Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\skjor\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\skjor\OneDrive\Documents\IT\_3D_Printer\_TwoTrees Sapphire Plus\Firmware\Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 262:
    load_marlin_features()
  File "C:\Users\skjor\OneDrive\Documents\IT\_3D_Printer\_TwoTrees Sapphire Plus\Firmware\Marlin-2.0.x_SapphirePlus_FromScratch_AutoAlignZ_FromVideoAuroraTech_ColorTouchscreen\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\skjor\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\skjor\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,

blinkitogaming, 2021-07-24:
Hi there, did you manage to solve this? Got this printer and I'd like to have this Marlin version working.

Thnx in advance!

mks-viva, 2021-08-09:
Maybe it is caused by the Vscode compilation environment.