makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #239: How to control fan?

RafaRS46, 2021-04-29:
Hello!
I conect a fan (layer fan) on PB1 port.
This fan is always on at full speed.
When I turn on the printer, the fan starts and I can't turn off it.

[image]

MKS Robin Nano V2

Thanks!

mks-viva, 2021-04-30:
Hi! You can use:
Open FAN: M106 S255   Close FAN: M107

RafaRS46, 2021-04-30:
Thanks but It doesn't work. :(
The fan conected on PB1 pin is always on. When I send M107, the fan keeps on. 
I don't understand what is wrong.

RicPaiva, 2021-05-02:
Hi @RafaRS46 
Not sure if this can help, but we had a very similar issue with a Robin E3P board and the problem was the MOSFET which controls the fan.
We solved the issue by replacing the SMD component.
Maybe you can have a look at it.
Good luck...

mks-viva, 2021-05-06:
@RafaRS46 I want to know, is the FAN turned on after you turn on the power when you have not control FAN?

RafaRS46, 2021-05-06:
Hi @mks-viva 
The fan turned on when I turned on the power supply.
The fan conected to PB1 never stops. 
In the image below, the indicated led os always on so I think that PB1 connector have always 24v.

[image]

Thanks

mks-viva, 2021-05-19:
@RafaRS46 Can you provide a video when your power on?

mks-viva, 2021-08-09:
I will close this issues.

closed 2021-08-09 by mks-viva