makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #245: [FR] Decouple LVGL and mks wifi

brdvvoz, 2021-05-10:
Right now it is only possible to use mks wifi with lvgl ui. Not everyone likes lvgl though, including me, enough that I don't use wifi solely because of that dislike.
So it would be great if the ability to use mks wifi independently from ui flavour used waas added.

mks-viva, 2021-05-19:
Hi! I have fed back your problem to the engineer.

Souls-Reaver, 2021-06-18:
También pido lo mismo, wifi independiente del perfil que se use, lo necesito para colour_ui. Gracias

cdExpat, 2022-02-09:
I have installed MKS Robin nano V 1.2 on a delta printer with MKS WiFi. LVGL ui does not support delta printers, so I need to use the standard Marlin interface. What do I have to do get MKS WiFi working.