makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #249: PID_AUTOTUNE_MENU

madurani, 2021-05-20:
Hello All,

I tried compile marlin with parameter "PID_AUTOTUNE_MENU". This parameter should be added to "Advanced Settings" menu the tool for auto-tuning of parameters Kp, Ki and Kd of HotEnd. Compilation was success but submenu wasn't added.
Could somebody help me mentioned issue?

Parameter "#define PID_AUTOTUNE_MENU" was uncoment but menu wasn't added:
```
// Comment the following line to disable PID and enable bang-bang.
#define PIDTEMP
#define BANG_MAX 255     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#define PID_K1 0.95      // Smoothing factor within any PID loop

#if ENABLED(PIDTEMP)
  #define PID_EDIT_MENU         // Add PID editing to the "Advanced Settings" menu. (~700 bytes of PROGMEM)
  #define PID_AUTOTUNE_MENU     // Add PID auto-tuning to the "Advanced Settings" menu. (~250 bytes of PROGMEM)
  //#define PID_PARAMS_PER_HOTEND // Uses separate PID parameters for each extruder (useful for mismatched extruders)
                                  // Set/get with gcode: M301 E[extruder number, 0-2]
```

argolomb, 2021-05-28:
Hi, having same problem here.

Chlorel, 2021-05-30:
the menu only appears with marlin UI

closed 2021-06-13 by madurani