makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #257: M500 >>>> Recv: echo:Error writing to EEPROM!

madurani, 2021-06-11:
Hello All
I use software Marlin v2 on MKS Robin Nano v2 hardware. Compilation and functionality of firmware work properly, only one issue is there. I can't store changed parameters sended from gcode console to EEPROM.
Every attempt to store parameters with command M500 finished with error. Could you please help me setup correctly parameters in file Configuration.h?

Example:
Attempt to change unit for x axis is success:
[...]
Send: M92 X99.506
Recv: ok
Attempt to store to changed parameters crashed:
Send: M500
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: ok
[...]

mks-viva, 2021-08-09:
If you send M92 X100 and it will work?

madurani, 2021-08-09:
Yes it work, value is changed.

mks-viva, 2021-08-10:
So, i will colose this issues.

madurani, 2021-08-10:
issue wasn't solded, command M500 finish with error.

mks-viva, 2021-08-11:
Plese provide your socrde code to me and i test it.

madurani, 2021-08-11:
Do you need completely source code or only some config files?

mks-viva, 2021-08-11:
source code

madurani, 2021-08-11:
Tarbal of source code has 108MB. Where I can store this package?

mks-viva, 2021-08-11:
You need delete .pio and upload *.zip to here
[image]

madurani, 2021-08-11:
After deleting of .pio folder, is zip file 27M, but from unknown reason I can't attach file here. Original source is 10M and is possible attache here.

madurani, 2021-08-11:
Tar comprimation is better. Now source have only 25MB and are in attachment:

[source1.tar.gz](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6968129/source1.tar.gz)

mks-viva, 2021-08-11:
Ok! I will test it tomorrow

madurani, 2021-08-11:
As gcode console I use console in Octoprint, but changed value isn't possible store nor via touch screen and icone "Store settings to eeprom". When is printer turn off , then all changed value are lost.

mks-viva, 2021-08-12:
I test it ok!
[image]
[image]

madurani, 2021-08-12:
I recompile firmware (add options #define EEPROM_BOOT_SILENT )  but problem persist.
[image]

Isn't needed set up some jumper on board?
Can by there some HW issue on board?

mks-viva, 2021-08-16:
You need not jump it, because eeprom is 3.3v power. You can try to send M503 and check your changes is ok?

madurani, 2021-08-16:
Zmeny zobrazené prostredníctvom M503 nie sú trvalé. Po vypnutí dosky MKS sa hodnoty zmenené prostredníctvom M92 stratia.
Znamená to, že príkaz M500 nevie zapísať zmenu do EEPROM a zmena zostane iba v pamäti RAM.

mks-viva, 2021-08-16:
Maybe your motherboard has a hardware problem, can you write eeprom to flash to ensure that the data can be saved

madurani, 2021-08-16:
Is there some "how to "/example, how can I safe data on flash?

mks-viva, 2021-08-16:
I need to test it before I can tell you how to set it up

mks-viva, 2021-08-17:
You can save date to sd card, like this set:
[image]

madurani, 2021-08-17:
And HW eeprom isn't needed disable in software/firmware?

madurani, 2021-08-17:
I tried it but implementation of firmware wasn't success

I uncommented code:
`#define SDCARD_EEPROM_EMULATION`
and commented:
```
/*
#if EITHER(NO_EEPROM_SELECTED, I2C_EEPROM)
  #define I2C_EEPROM                              // EEPROM on I2C-0
  #define MARLIN_EEPROM_SIZE              0x1000  // 4KB
#endif
*/
```
Compilation of Marlin was success, but implementation of firmware to MKS board from SD-card crashed:
First screen  "TFT updating..." was OK, but second "Assets Updating..." wasn't appear, screen stay black.

After that i tried to commented #define SDCARD_EEPROM_EMULATION and uncommented  #define FLASH_EEPROM_EMULATION and downloading firmware and functionality of command M500 work correctly.
Values after power off MKS board are persistent store.

Send: M92 X100
[...]
Recv: ok
[...]
Send: M500
Recv: ok
[...]
Send: M501
...
...
...
Recv: echo: M92 X100.00 Y100.00 Z400.00 E431.35

**Can I use as EEPROM flash memory?**

mks-viva, 2021-08-17:
No, because i has test it and can not work. So, you only save eeprom data to sd card.

madurani, 2021-08-17:
What not work? When I active flash as eeprom, then I can store values with M500 and after turn off/on board are all changed values persistent store.

mks-viva, 2021-08-17:
The board will always restart. So, save eeprom data to sd card.

madurani, 2021-08-17:
When I active sd card as eeprom, then I can't install firmware to MKS board. Screen stay black.

mks-viva, 2021-08-17:
Can you provide a pictrue to me about board's EEPROM chip. It's AT24C32DM.
[image]

madurani, 2021-08-17:
Here is foro:
[image]

mks-viva, 2021-08-18:
16CM is 2k eeprom, the marlin firmware need >= 4k. Can you change to AT24C32DM by yourself?

madurani, 2021-08-18:
I need more instructions what is needed to change.

mks-viva, 2021-08-18:
You need remove your board's eeprom chip and welded a AT24C32DM chip.

madurani, 2021-08-18:
Do you mean physically switch chips on board?

mks-viva, 2021-08-18:
Yes

madurani, 2021-08-18:
and isn't possible set up smaller eeprom size, from 4K to 2K?
#define MARLIN_EEPROM_SIZE              0x800  // 2KB

mks-viva, 2021-08-18:
In my memory, EEPROM needs to be ≥4K

mks-viva, 2021-08-24:
I will close this issues temporarily.

closed 2021-08-24 by mks-viva