makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #260: More Menus do not show items

FabioMC, 2021-06-20:
After I compiled the last firmware the "More" menu on "Tools" only presents a icon called G-Code. All the others icons like preheat PLA, preheat ABS,  and all the others configured on configuration_adv is not present.

[image]

This is how my configuration_adv is right now.
// Custom Menu: Main Menu
#define CUSTOM_MENU_MAIN
#if ENABLED(CUSTOM_MENU_MAIN)
  #define CUSTOM_MENU_MAIN_TITLE "Custom Commands"
  #define CUSTOM_MENU_MAIN_SCRIPT_DONE "M117 User Script Done"
  #define CUSTOM_MENU_MAIN_SCRIPT_AUDIBLE_FEEDBACK
  #define CUSTOM_MENU_MAIN_SCRIPT_RETURN   // Return to status screen after a script
  #define CUSTOM_MENU_MAIN_ONLY_IDLE         // Only show custom menu when the machine is idle

  #define MAIN_MENU_ITEM_1_DESC "Home & UBL Info"
  #define MAIN_MENU_ITEM_1_GCODE "G28\nG29 W"
  //#define MAIN_MENU_ITEM_1_CONFIRM          // Show a confirmation dialog before this action

  #define MAIN_MENU_ITEM_2_DESC "Preheat for " PREHEAT_1_LABEL
  #define MAIN_MENU_ITEM_2_GCODE "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)
  //#define MAIN_MENU_ITEM_2_CONFIRM

  #define MAIN_MENU_ITEM_3_DESC "Preheat for " PREHEAT_2_LABEL
  #define MAIN_MENU_ITEM_3_GCODE "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)
  //#define MAIN_MENU_ITEM_3_CONFIRM

  //#define MAIN_MENU_ITEM_4_DESC "Heat Bed/Home/Level"
  //#define MAIN_MENU_ITEM_4_GCODE "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"
  //#define MAIN_MENU_ITEM_4_CONFIRM

  //#define MAIN_MENU_ITEM_5_DESC "Home & Info"
  //#define MAIN_MENU_ITEM_5_GCODE "G28\nM503"
  //#define MAIN_MENU_ITEM_5_CONFIRM
#endif

Valnar8, 2021-06-21:
there is already a thread for this #21423

Valnar8, 2021-06-21:
There is already a thread for this 
https://github.com/MarlinFirmware/Marlin/issues/21423#issue-838868707

javicarrera, 2021-07-30:
I think I have found the reason for this problem.
With this commit #18177 definitions for custom user menus were changed in Marlin.

#define CUSTOM_USER_MENUS
#define USER_DESC_1 ""
#define USER_GCODE_1 ""
was changed by:
#define CUSTOM_MENU_MAIN
#define MAIN_MENU_ITEM_1_DESC ""
#define MAIN_MENU_ITEM_1_GCODE ""

It is missing however changing those in conditionals_adv.h (CUSTOM_MENU_MAIN was changed, the other 2 not):
#if BOTH(HAS_TFT_LVGL_UI, CUSTOM_MENU_MAIN)
#define HAS_1(N) (defined(USER_DESC##N) && defined(USER_GCODE_##N))

Please if someone can change this line of code and I think the "more" menu will work again with LVGL UI.

javicarrera, 2021-07-31:
In today's Marlin bugfix it is solved!