makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #261: MKS_WIFI_MODULE + ESP3D

iNDeX999, 2021-06-25:
Hi,

i have flashed ESP3D to my MKS WIFI 
(Firmware: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware.git)

NANO V3 / TS35
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#define TFT_LVGL_UI
#define MKS_WIFI_MODULE

ESP3D FW_VERSION "2.1.1.b10" with and without MKS_TFT_FEATURE compiled
)

But i cannot connect to the printer in the ESP3D Gui

Did i misse something?

DvoraNoob, 2021-06-26:
> Hi,
>
> i have flashed ESP3D to my MKS WIFI
> (Firmware: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware.git)
>
> NANO V3 / TS35
> #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
> #define [...]

esp 3.0 , not 2.1.1 
https://github.com/luc-github/ESP3D/tree/3.0

iNDeX999, 2021-06-28:
Ok, connection works now. 
But... it´s more or less useless, thx MKS.  
How can i use ESP3D as im used to from my fysetc e4.
Use an ESP32 and connect it to power +RX/TX from the Wifi socket off the nano v3?

DvoraNoob, 2021-06-28:
> Ok, connection works now.
> But... it´s more or less useless, thx MKS.
> How can i use ESP3D as im used to from my fysetc e4.
> Use an ESP32 and connect it to power +RX/TX from the Wifi socket off the nano v3?

ESP you can only use in web