makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #263: REPRAP_DISCOUNT_SMART_CONTROLLER

seymon92, 2021-06-29:
Hello,
I have problem with my old Reprap controller, if I enable this display (#define REPRAP_DISCOUNT_SMART_CONTROLLER) in Configuration.h display only lights up and thats all. I am using board MKS Robin nano V3.

Can somebody help me?

alexandremstf, 2021-07-03:
Check the orientation of your flat cable, some times they don't match with the boards connection, and you will need to cut off the cable connector to allow rotate and insert in another orientation.

seymon92, 2021-07-07:
Hi,
Thank You for reply, but connectors are correct (already rotated).

mks-viva, 2021-07-07:
You only enable #define REPRAP_DISCOUNT_SMART_CONTROLLER and connect it is ok

seymon92, 2021-07-13:
If I enable #define REPRAP_DISCOUNT_SMART_CONTROLLER I get this error:
'boolean' is deprecated [-Wdeprecated-declarations]

There is problem in these files:
Marlin-2.0.x\.pio\libdeps\mks_robin_nano_v3_usb_flash_drive_msc\LiquidCrystal\FastIO.cpp
Marlin-2.0.x\.pio\libdeps\mks_robin_nano_v3_usb_flash_drive_msc\LiquidCrystal\FastIO.h

alexandremstf, 2021-07-21:
I have the same problem, and i can't solve. Now i am using mks ts35.

mks-viva, 2021-08-07:
Use #if IS_ULTIPANEL replace #if ENABLED(ULTIPANEL)
@seymon92 
[image]

closed 2021-08-07 by mks-viva