makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #267: Black screen after auto leveling

seba1213, 2021-07-12:
I merged config files from old version (maybe one year before) witch curent version. ven i select an option auto leveling on screen after leveling screen become black, only reboot help. I use an inductive proximity sensor, attaching my configs.
[Configuration_adv.h.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6804162/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6804163/Configuration.h.txt)

siwanamu, 2021-07-20:
It's a firmware bug.
In case of FIX_MOUNTED_PROBE, a black screen occurs.

This is a temporary workaround.

draw_level_settings.cpp
1. Line 65 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

2. Line 90 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

draw_ui.cpp
1. Line 904 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

2. Line 1001 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)
  

printer_operation.cpp
1. Line 170~173 : Add

168:        lv_draw_touchmi_settings();
169:      #endif
170:      #ifdef FIX_MOUNTED_PROBE
171:        bltouch_do_init(false);
172:        lv_draw_bltouch_settings();
173:      #endif
174:      uiCfg.autoLeveling = false;

siwanamu, 2021-07-20:
It's a firmware bug???
In case of FIX_MOUNTED_PROBE, a black screen seems to have occurred.

This is a temporary workaround.

draw_level_settings.cpp
1. Line 65 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

2. Line 90 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

draw_ui.cpp
1. Line 904 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)

2. Line 1001 : Modify
#if ENABLED(BLTOUCH)
  --> #if ENABLED(BLTOUCH) || ENABLED(FIX_MOUNTED_PROBE)
  

printer_operation.cpp
1. Line 163
#ifdef BLTOUCH
  --> #ifdef BLTOUCH || FIX_MOUNTED_PROBE

karabirkan, 2023-03-26:
Same thing happened and even there is power on 3dtouch, mainboard etc, display is blank, no beep sound, not turning on and it does not update firmware from sd card...