makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #268: sensors homming

DoctorDeatch, 2021-07-26:
Hi friends I have a problem, the board does not want to see home sensors, it just doesn’t react to them.
What could be problematic?

mks-viva, 2021-07-31:
You can refer to here: https://github.com/makerbase-mks/MKS-Robin-Nano-V3.X/wiki/Drivers_MKS_TMC2209_2225_2208

DoctorDeatch, 2021-07-31:
I do not want to use unloading tms
i want to use limit switches

mks-viva, 2021-07-31:
You doesn't enable SENSORLESS_HOMING

DoctorDeatch, 2021-07-31:
As far as I understand SENSORLESS HOMING is necessary for TMS, based on engine loads. I want to use limit switches.

DoctorDeatch, 2021-07-31:
sensor h also does not work

mks-viva, 2021-08-07:
May be you need set the value of ENDSTOP_INVERTING in configuration.h file

madurani, 2021-08-13:
On this link is how to for sensor-less-homing for this HW:  https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/Drivers_MKS_TMC2209_2225_2208

Warning: tested on tmc2209v2 for coreXY

I configured it exactly according to "how to" and it worked for me straight on the first time, only values of X_STALL_SENSITIVITY Y_STALL_SENSITIVITY I set up on higher than  recommended >> 170,  because bump was too hard.

closed 2021-08-17 by mks-viva