makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #269: my mini12864 v3 not working on robin nano v3

Wayne-Weng, 2021-07-31:
### Description

I want to use mks robin nano v3 with mks mini12864 v3 controller. but there's no #define MKS_MINI_12864_V3 in the configuration.h file, only there is the #define MKS_MINI_12864. So I enable #define MKS_MINI_12864, along with #define NEOPIXEL_LED. After firmware was built,  I copied firmware.bin(and renamed to robin_nano_v3.bin) and assets folder to sd card and reboot the motherboard. The mks mini 12864 controller v3 shows nothing. Just black screen. The beeper works though.

I have commented out the #define MKS_TS35_V2_0, #define TFT_LVGL_UI and #define TOUCH_SCREEN because I don't want to use TS35.
I use the latest bugfix version from this repository, my configuration.h version is 020008.

**Expected behavior:** The mks mini 12864 v3 should show boot screen and menu.
**Actual behavior:** screen shows nothing

I've attached the config files.
[Marlin.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6910958/Marlin.zip)

mks-viva, 2021-07-31:
You need download the latest marlin bugfix-2.0.x

Wayne-Weng, 2021-07-31:
Hi I just replaced /Marlin folder with the latest /Marlin folder from the original author, the other folder like /buildroot and /ini are unchanged(Because I want to avoid messing with boot and firmware config sort of things), and it works, thanks.

mks-viva, 2021-07-31:
I will close this issues

closed 2021-07-31 by mks-viva