makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #272: Z-max sensor less homing on MKS Robin Nano v2

madurani, 2021-08-20:
Hello MKS Support,

I have configured on board **MKS Robin Nano v2** sensor less homing on axis xy(driver TMC2209) and on axis z as Zmin bltouch. All work properly and correctly.
Now I want to try configure sensor zmax(driver tmc2226) as sensor less homing.
**Is possible active zmax via sensor less homing together with bltouch on zmin?**

**What should be enable in marlin?**
 **Configuration.h**
#define USE_YMAX_PLUG
Z_MAX_ENDSTOP_INVERTING false
**Configuration_adv.h**
#define Z_STALL_SENSITIVITY  200

**I want detect zmax value via zmax sensor. Is it posible?**

mks-viva, 2021-08-21:
I don’t recommend the Z axis to use the sensor less homing function, usually XY use it.

madurani, 2021-08-21:
Is there some **how to** for installation of **MKS EXP-MOT3 UART** and configuration in marlin?  I need next driver for second z-motor, because I want to try "Align Z stepper motors". Both extruders drivers on HW Robin Nano v2 I have in use.

closed 2021-08-24 by madurani