makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #273: Can't adjust the offset Z parameter

DoctorDeatch, 2021-08-21:
Can't adjust the alignment sensor
the system does not allow the table to be moved to -Z
the set parameter is d EEPROM Z probe offset ignored both negative and positive 
how to overcome the problem?

madurani, 2021-08-21:
Which HW do you use?  
I use **MKS Robin Nano v2** and I have an same issue: storing to eeprom.
Problem is, that on board is only 2KB eeprom(should be 4KB) and Marlin needs eeprom 4KB.
 I solved problem via printing in **OctoPrint**. I have this parameter **offset Z** stored in **"GCODE Scripts/Before print job starts"**.
In the 6th line. Number  Z-0.93.  I do want to easily rewrite, when it is needed.

```
G21 ;metricka soustava pro hodnoty
G90 ;pozicovani absolutnimi hodnotami
G28 X0 Y0 ;presun X/Y na min endstops
G28 Z0 ;presun Z na min endstops
G29 ; AutoLevel
M851 X-24.0 Y-0.0 Z-0.93
M82 ;
```

DoctorDeatch, 2021-08-21:
I have an MKS Robin Nano v3 but there is also 4k  
but I do not use engine load stop, I have xy contact 

also besides this, I cannot move the table to minus
that is, I cannot calibrate the table according to the sensor in the negative direction at all

madurani, 2021-08-21:
I use for xy sensor less homing and for z blotouch.
Auto bed leveling and offset z work for me fine in marlin.

DoctorDeatch, 2021-08-21:
I have an induction sensor

mks-viva, 2021-08-25:
@DoctorDeatch 
Enable '#define EEPROM_SETTINGS' in Configuration.h file?