makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #274: EXP-MOT3 and Robin Nano v2

madurani, 2021-08-21:
What if I want to use uart mode(HW Robin Nano v2)? How will it be with CS and EN port, since they are on the same pin on scheme.
For example when I need to add second Z2-driver on M1, then I add in pin file:
**pins_MKS_ROBIN_NANO_V2.h**
```
#define Z2_ENABLE_PIN                        PE13
#define Z2_STEP_PIN                          PD13
#define Z2_DIR_PIN                           PC6
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                          PE13
#endif
```
How will it be with display? Options **#define TFT_LVGL_UI** and **#define TOUCH_SCREEN?** Should be commented in config file?
Or is there any other possibility to use display  **MKS TS35**?
BTW: Display for me isn't so important, because I print from OctoPrint via usb port.

_Originally posted by @madurani in https://github.com/makerbase-mks/MKS-Robin-E3-E3D/issues/125#issuecomment-903173795_

mks-viva, 2021-08-23:
The MKS EXP-MOT3 board connect to motherboard by EXP1,EPX2, so, you only select MKS TS35 or MKS EXP-MOT3

madurani, 2021-08-23:
What will be with collision CS and EN ports? They are on same pin.

_What if I want to use uart mode(HW Robin Nano v2)? How will it be with CS and EN port, since they are on the same pin on scheme.
For example when I need to add second Z2-driver on M1, then I add in pin file:
pins_MKS_ROBIN_NANO_V2.h_
```
#define Z2_ENABLE_PIN                        PE13
#define Z2_STEP_PIN                          PD13
#define Z2_DIR_PIN                           PC6
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                          PE13
#endif
```

mks-viva, 2021-08-25:
MKS TS35 and EXP-MOT3 cann't be use on MKS Robin Nano V2 motherboard at the same time.

madurani, 2021-08-25:
OK I understand, that display cannot be used together with EXP-MOT3. And what about EN pins? There are 3 motors and only one EN(enable) pin. All motors will be configured on same EN pin?

mks-viva, 2021-08-25:
Yes, EXP-MOT3 all use the same EN  pin.

madurani, 2021-08-25:
Is possible try to join EXP-MOT3 to same free ports on WIFI, THM, MT_DET, Z-max and etc and no to EXP1/2?

mks-viva, 2021-08-25:
Your idea is to connect WIFI, THM, MT_DET, Z-max signals to the EXP-MOT3 drive motor?

madurani, 2021-08-25:
Yes, free unused pins/ports from mentioned sockets, connect to EXP-MOT3 + 5V/24V

mks-viva, 2021-08-25:
I have a temporary method, you can use EXP-MOT3 and TS35 at the same time. MKS EXP-MOT3's EXP1 and EXP2 signal connect to WIFI, MT_DET and so on signal interface

madurani, 2021-08-25:
OK, I will test it, but I am waiting on delivery of equipment EXP-MOT3. Please don't close this ticket.

mks-viva, 2021-08-25:
Ok, after your test is complete, I am closing this issue.

madurani, 2021-09-10:
Equipment EXP-MOT3 was delivered, but did not work for me.
Goal was to add one z-stepper-motor(Z2) and leave connected display on HW MKSRobiNanoV.2

One motor with uart mode of control requires 4 ports/pins(enable,step,dir,uart)
I used first socket of equipment EXP1 (pins: EN, STEP1, DIR1, UART1, +5V, GND), power socket DC24V+/- and connected it to robin board. I tried different ports. Response on command M122 for motor Z2 is positive, but motor installed to socket M1 has not rotated.
Used ports:
EN=PA10
STEP1=PA9
DIR1=PC7
UART1=PC13
Probably some of ports aren't suitable for motor configuration. Can you advice any  more suitable ports?

[image]

[image]

[image]

 
```
Send: M122
Recv: 		        X	Y	Z	Z2	E       E1
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false	false	false
Recv: Set current	800	800	1600	800	800	800
Recv: RMS current	795	795	1546	795	795	795
Recv: MAX current	1121	1121	2180	1121	1121	1121
Recv: Run current	25/31	25/31	27/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	13/31	12/31	12/31	12/31
Recv: CS actual	12/31	12/31	13/31	12/31	12/31	12/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	0=.325	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true	true
Recv: msteps		16	16	16	16	16	16
Recv: interp		true	true	true	true	true	true
Recv: tstep		max	max	max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false	false
Recv: pwm scale sum	14	14	15	14	14	14
Recv: pwm scale auto	0	0	0	0	0	0
Recv: pwm offset auto	36	36	36	36	36	36
Recv: pwm grad auto	14	14	14	14	14	14
Recv: off time	4	4	4	4	4	4
Recv: blank time	24	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2	2
Recv:  -start		1	1	1	1	1	1
Recv: Stallguard thrs	170	170	0	0
Recv: uStep count	40	24	1000	984	40	24
Recv: DRVSTATUS	X	Y	Z	Z2	E	E1
Recv: sg_result	0	0	0	0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0D:00:00
Recv: 		Z2	0xC0:0C:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 		E1	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok
```

**Marlin Configuration:**

**pins_MKS_ROBIN_NANO_V2.h**
```
/Z2
#define Z2_ENABLE_PIN                        PA10
#define Z2_STEP_PIN                          PA9
#define Z2_DIR_PIN                           PC7
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                          PC13
#endif

 #define Z2_SERIAL_TX_PIN                  PC13
 #define Z2_SERIAL_RX_PIN                  PC13
```
**Configuration_adv.h**
```
#define NUM_Z_STEPPER_DRIVERS 2 

#if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT      800
    #define Z2_CURRENT_HOME Z2_CURRENT
    #define Z2_MICROSTEPS    Z_MICROSTEPS
    #define Z2_RSENSE         0.11
    #define Z2_CHAIN_POS     -1
    //#define Z2_INTERPOLATE true
  #endif
```

**Configuration.h**
`#define Z2_DRIVER_TYPE TMC2209
`

mks-viva, 2021-09-11:
[image]
This means that the UART can work. 
M1 PIN map：
[image]

madurani, 2021-09-11:
Yes, I used wifi pins, but motor not rotated. Do You have some idea what can be wrong?

**What were done:**
I created socket( which was plugged in wifi port on robin board) . These are cable description:
**green**   >> PC13(_on scheme is mistake, there is P13 and should be PC13_) >> UART1
**yellow**  >> PC7 >> DIR1
**orange** >> PA9 >> STEP1
**white**   >> PA10 >> EN
**Detail of socket:**
[image]
**Socket plugged to fifi port:**
[image]
**Here is second side of cable plugged to equipment EXP-MOT3 in port EXP1 for motor M1:**
[image]
**Here is detail from pins_MKS_ROBIN_NANO_V2.h**
[image]
**Output from M122 looks fine but motor Z2 does not work:**
```
Recv: ok
[...]
Send: M122
Recv: 		        X	Y	Z	Z2	E	E1
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false	false	false
Recv: Set current	800	800	1600	800	800	800
Recv: RMS current	795	795	1546	795	795	795
Recv: MAX current	1121	1121	2180	1121	1121	1121
Recv: Run current	25/31	25/31	27/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	13/31	12/31	12/31	12/31
Recv: CS actual	12/31	12/31	13/31	12/31	12/31	12/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	0=.325	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true	true
Recv: msteps		16	16	16	16	16	16
Recv: interp		true	true	true	true	true	true
Recv: tstep		max	max	max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false	false
Recv: pwm scale sum	14	14	15	14	14	14
Recv: pwm scale auto	0	0	0	0	0	0
Recv: pwm offset auto	36	36	36	36	36	36
Recv: pwm grad auto	14	14	14	14	14	14
Recv: off time	4	4	4	4	4	4
Recv: blank time	24	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2	2
Recv:  -start		1	1	1	1	1	1
Recv: Stallguard thrs	170	170	0	0
Recv: uStep count	24	40	1000	984	24	24
Recv: DRVSTATUS	X	Y	Z	Z2	E	E1
Recv: sg_result	0	0	0	0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0D:00:00
Recv: 		Z2	0xC0:0C:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 		E1	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK
Recv: ok
[...]
Send: G91
Recv: ok
Send: G1 Z-10 F200
Recv: ok
Send: G90
Recv: ok
```

madurani, 2021-09-12:
It works!!! I found happy combinations of pins. I tried plenty combinations of ports(problem is that i don't have knowledge background related to functionality of ports ). 
Now command G34 is working with two z-motor and display stays installed.
EN     >> PC4(Z+)
STEP >> PC7(wifi IO1)
DIR   >> PB2()
UART >> PC13(wifi IO0)
But chances, that I can find next three suitable ports for third Z3 motors are very slim.  
```
[...]
Send: G34
Recv: 
Recv: G34 Iteration: 1
Recv: DIFFERENCE Z1-Z2=0.07
Recv: G34 Iteration: 2
Recv: DIFFERENCE Z1-Z2=0.12
Recv: Decreasing Accuracy Detected.
Recv: G34 Iteration: 3
Recv: DIFFERENCE Z1-Z2=0.08
Recv: G34 Iteration: 4
Recv: echo:busy: processing
Recv: DIFFERENCE Z1-Z2=0.03
Recv: G34 Iteration: 5
Recv: DIFFERENCE Z1-Z2=0.03
Recv: Did 5 of 5
Recv: Accuracy: 0.03
```
[image]

mks-viva, 2021-09-13:
Maybe only one Z axis can be added. This is also good news, at least it can work.

madurani, 2021-09-13:
I don't know, but it look like in marlin can be defined four z-motors. Please not close this ticket I will try add third motor. 

Example from marlin
```
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
//#define Z3_DRIVER_TYPE TMC2209
//#define Z4_DRIVER_TYPE A4988

// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 1   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  // Enable if Z motor direction signals are the opposite of Z1
  //#define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR

```

madurani, 2021-09-14:
It look like, that i fond ports for third z3-motor. For now only on driver A4988 on step/dir mode, but today I will test it with driver tmc2226 on uart mode. Display is still installed.

```
 //Z2

#define Z2_ENABLE_PIN                        PC4
#define Z2_STEP_PIN                          PC7
#define Z2_DIR_PIN                           PB2 

//Z3
#define Z3_ENABLE_PIN                        PC4
#define Z3_STEP_PIN                          PE5 
#define Z3_DIR_PIN                           PA2 
```

madurani, 2021-09-14:
Driver tmv2226 was added. Uart configured. It look like motor Z3 work:

```
Send: M122
Recv: 		        X	Y	Z	Z2	Z3	E	E1
Recv: Address		0	0	0	0	0
Recv: Enabled		false	false	false	false	false	false	false
Recv: Set current	800	800	1600	800	800	800	800
Recv: RMS current	795	795	1546	795	795	795	795
Recv: MAX current	1121	1121	2180	1121	1121	1121	1121
Recv: Run current	25/31	25/31	27/31	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	13/31	12/31	12/31	12/31	12/31
Recv: CS actual	12/31	12/31	13/31	12/31	12/31	12/31	12/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	0=.325	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true	true	true
Recv: msteps		16	16	16	16	16	16	16
Recv: interp		true	true	true	true	true	true	true
Recv: tstep		max	max	max	max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false	false	false
Recv: pwm scale sum	14	14	15	30	69	14	14
Recv: pwm scale auto	0	0	0	16	55	0	0
Recv: pwm offset auto	36	36	36	36	36	36	36
Recv: pwm grad auto	14	14	14	14	14	14	14
Recv: off time	4	4	4	4	4	4	4
Recv: blank time	24	24	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2	2	2
Recv:  -start		1	1	1	1	1	1	1
Recv: Stallguard thrs	170	170	0	0	0
Recv: uStep count	40	40	1000	24	1000	40	24
Recv: DRVSTATUS	X	Y	Z	Z2	Z3	E	E1
Recv: sg_result	0	0	0	0	0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0D:00:00
Recv: 		Z2	0xC0:0C:00:00
Recv: 		Z3	0xC0:0C:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 		E1	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing Z3 connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK

```

madurani, 2021-09-19:
I added the fourth driver/stepper(display was unmounted) and now I have aligning via four independent steppers. There was trouble with MKS marlin firmware, because is too old v2.0.5.3. When I matched the source code for command G34 (Marlin/src/gcode/calibrate/G34_M422.cpp) I found out, that there are big differences compared to the last version of marlin 2.0.9.1. 
On an old firmware G34 for quadruple z-steppers was working horribly. After compilation of the last version of Marlin, G34 works much better.

```
Recv: Z2-Z1=0.02 Z3-Z2=0.02 Z3-Z1=0.04 Z4-Z3=0.06 Z4-Z2=0.04 Z4-Z1=0.0
Recv: Did 10 of 10
Recv: Accuracy: 0.06

```

 
[image]
[image]
[image]
[image]

madurani, 2021-09-22:
Hello MKS support,

Could you please advise me with some secondary issues on EXP-MOT3 equipment ?

1. During loading of new firmware the board begins to buzz loudly. My solution is unplug EXP1 socket during loading new firmware. Is it possible to somehow deactivate this buzzing ? 
2. After the start of the printer, the board beeps shortly and Z-motors (Z2,Z3, and Z4), that are plugged on EXP-MOT,  rotate a bit randomly.  As result the angle of the plane of hotbed is lost.

madurani, 2021-11-04:
First attempt connect second EXP-MOT3 equipment(with two drivers without motors) look fine: Four driver for Z-motors and Four drivers for Extruders

[image]

```
Send: M122
Recv: 		        X	Y	Z	Z2	Z3	Z4	E	E1	E2	E3
Recv: Address		0	0	0	0	0	0			0	0
Recv: Enabled		false	false	false	false	false	false	false	false	false	false
Recv: Set current	800	800	750	750	750	750	800	800	600	600
Recv: RMS current	795	795	734	734	734	734	795	795	581	581
Recv: MAX current	1121	1121	1035	1035	1035	1035	1121	1121	819	819
Recv: Run current	25/31	25/31	23/31	23/31	23/31	23/31	25/31	25/31	18/31	18/31
Recv: Hold current	12/31	12/31	11/31	11/31	11/31	11/31	12/31	12/31	9/31	9/31
Recv: CS actual	12/31	12/31	11/31	11/31	11/31	11/31	12/31	12/31	9/31	9/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18	1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true	true	true	true	true	true
Recv: msteps		16	16	16	16	16	16	16	16	16	16
Recv: interp		true	true	true	true	true	true	true	true	true	true
Recv: tstep		max	max	max	max	max	max	max	max	max	max
Recv: PWM thresh.	79	79	658	658	658	658	61	61	61	61
Recv: [mm/s]		100	100	3	3	3	3	30	30	30	30
Recv: OT prewarn	false	false	false	false	false	false	false	false	false	false
Recv: pwm scale sum	14	14	13	33	28	42	14	14	62	62
Recv: pwm scale auto	0	0	0	20	15	29	0	0	51	51
Recv: pwm offset auto	36	36	36	36	36	36	36	36	36	36
Recv: pwm grad auto	14	14	14	14	14	14	14	14	14	14
Recv: off time	4	4	4	4	4	4	4	4	4	4
Recv: blank time	24	24	24	24	24	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2	2	2	2	2	2
Recv:  -start		1	1	1	1	1	1	1	1	1	1
Recv: Stallguard thrs	140	140	0	0	0	0			0	0
Recv: uStep count	40	40	8	1000	8	296	40	8	40	8
Recv: DRVSTATUS	X	Y	Z	Z2	Z3	Z4	E	E1	E2	E3
Recv: sg_result	0	0	0	0	0	0			0	0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0xC0:0C:00:00
Recv: 		Z	0xC0:0B:00:00
Recv: 		Z2	0xC0:0B:00:00
Recv: 		Z3	0xC0:0B:00:00
Recv: 		Z4	0xC0:0B:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 		E1	0xC0:0C:00:00
Recv: 		E2	0xC0:09:00:00
Recv: 		E3	0xC0:09:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing Z3 connection... OK
Recv: Testing Z4 connection... OK
Recv: Testing E connection... OK
Recv: Testing E1 connection... OK

```

mks-viva, 2021-12-17:
So, i will close this issues temporarily.

closed 2021-12-17 by mks-viva