makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #275: Scrolling g-code LCD output

javicarrera, 2021-08-23:
### Description
When sending gcode command on LVGL interface, if the output message is too long it is just truncated.
For example when doing PID autotune throught the LCD itself you can see only the first lines, but not the actual 
PID values at the end of the command line.

Is it possible to include some kind of scrolling for the gcode output? 
Let say : button up-down or next-previous
I know that can be done with Pronterface but it doesn't make sense to me to connect to a computer every time you want to set something, in particular if there is no computer close by.

#### Additional Information

Picture showing the problem:
http://prntscr.com/1qfegkk

mks-viva, 2021-08-25:
@solawc Please check it.

solawc, 2021-08-25:
> ### Description
> When sending gcode command on LVGL interface, if the output message is too long it is just truncated.
> For example when doing PID autotune throught the LCD itself you can see only the [...]

This is a good idea. It is similar to the current file name display. If it is too long, the display will be incomplete. Therefore, I am modifying the display effect of this part. I intend to turn it into a one-way scrolling form. Please pay attention to the next version update.