makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #278: Missing icons after last firmware modification

FabioMC, 2021-09-15:
After the last update done by @solawc the icons on different pages disapears. I alreadt tried to erase assets folder and let VS to download it again but no fix. 

Any hint?

[image]
[image]
[image]
[image]

mks-viva, 2021-09-16:
Please update asset.

solawc, 2021-09-16:
> assets

Hello, you try to download the last version of the firmware without updating the assets folder to see if the display is normal

solawc, 2021-09-16:
> After the last update done by @solawc the icons on different pages disapears. I alreadt tried to erase assets folder and let VS to download it again but no fix.
> 
> Any hint?
> 
> [image]
> [image]
> [image]
> [image]

Hello, you try to download the last version of the firmware without updating the assets folder to see if the display is normal

FabioMC, 2021-09-16:
In fact no. As I said, I erased the assets folder and let visual code to download a new assets folder but the result was the same. I will unistall all and try a new clean setup.