makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #279: 'MT_TIME_DELAY' was not declared in this  scope

pforrmi, 2021-09-17:
Hi all.

I want to compile the firmware for my sapphire plus with 1.2 board.
I double checked the preconfiguration which is needed and not able to figure out where is the problem.

`
Marlin/src/lcd/extui/lib/mks_ui/printer_operation.cpp:191:27: error: 'MT_TIME_DELAY' was not declared in this
 scope
   filament_c.tick_delay = MT_TIME_DELAY;
                           ^~~~~~~~~~~~~
*** [.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/printer_operation.cpp.o] Error 1
========================= [FAILED] Took 112.22 seconds =========================`

Maybe you can have a look or give me a hint.

Thanks

[Archiv.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7184554/Archiv.zip)
!

MikeGroovy, 2021-09-17:
I was able to overcome this error by editing
Mks-Robin-Nano-Marlin2.0-Firmware-master\Marlin\src\lcd\extui\lib\mks_ui\printer_operation.cpp

Changed 
`filament_c.tick_delay =  MT_TIME_DELAY`
to 
`filament_c.tick_delay = 3000; `

3000 or 3seconds is what MT_TIME_DELAY is set to. 

In other news I'm using a Sainsmart Coreception 300 (MKS Robin Nano 1.2) but after I compiled and flashed I get a message that "Printer Halted please reset"
I'm disabling things bit by bit to see if I can isolate the cause.

Been using a slightly modified version of the firmware below with no problem but was really hoping to try out the LVGL features :D
https://github.com/jgarder/Marlin-Coreception-300

MikeGroovy, 2021-09-17:
Found that Mks-Robin-Nano-Marlin2.0-Firmware-master\Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO.h
was missing a line for defining MT_TIME_DELAY

Marlin\src\pins\stm32f4\pins_MKS_ROBIN_NANO_V3.h had it defined though. 
added it under // Misc. Functions area..
`#define MT_TIME_DELAY 3000  // Default 3s  `

Then set Mks-Robin-Nano-Marlin2.0-Firmware-master\Marlin\src\lcd\extui\lib\mks_ui\printer_operation.cpp back to 
`filament_c.tick_delay = MT_TIME_DELAY`

Compiled with no error.

pforrmi, 2021-09-18:
Thanks very much! What a great and quick fix 👍

But I got another issue with this firmware. My X and Y are reversed.
If I want to control X the Y Axis moves and controlling the Y the X Axis moves. The connection onto the mainboard is correct. 
I think just a little change in marlin is needed but I can't find anything.

Maybe you can help me with this, too?!

pforrmi, 2021-09-18:
Alright, I inverted the steppers and now it works fine.
Thanks for the help.

Closed.

closed 2021-09-18 by pforrmi