makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #282: Robin nano V1.2 "MT_TIME_DELAY" not declared

Davide655321, 2021-09-29:
[image]

phcay, 2021-10-05:
Yes, I have same problem. MT_TIME_DELAY missing in pins_MKS_ROBIN_NANO.h and pins_MKS_ROBIN_NANO_V2.h. Present in pins_MKS_ROBIN_NANO_V3.h

\Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO.h
...
  #define MT_DET_1_PIN                      PA4   // LVGL UI FILAMENT RUNOUT1 PIN
  #define MT_DET_2_PIN                      PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define MT_DET_PIN_INVERTING             false  // LVGL UI filament RUNOUT PIN STATE
  //// PhC : MT_TIME_DELAY missing in pins_MKS_ROBIN_NANO.h and pins_MKS_ROBIN_NANO_V2.h. Present in pins_MKS_ROBIN_NANO_V3.h :
  #define MT_TIME_DELAY                     3000  // Default 3s   
...

solawc, 2021-10-22:
> Yes, I have same problem. MT_TIME_DELAY missing in pins_MKS_ROBIN_NANO.h and pins_MKS_ROBIN_NANO_V2.h. Present in pins_MKS_ROBIN_NANO_V3.h
>
> \Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO.h ... #define [...]

The new version does not have this problem, because the macro was only applied to NanoV3 at the time, and it was later modified.