makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #284: probleme 3dtouch

steph5764, 2021-10-09:
bonjour , j'ai un possède une mks nano v3 j'utilise le firmware marlin et je souhaite utiliser un 3dtouch mais je rencontre un probleme lors du z homing le premier palpage fonctionne les moteur z s'arrêtent puis le palpeur tente une deuxieme foi et là le 3dtouch clignotte  ?? 
merci d'avance pour vos aide
[Configuration.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7316316/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7316317/Configuration_adv.txt)
merci

phcay, 2021-10-10:
Je n'ai rien vu de déconnant dans ta conf sur le BLTOUCH. Ma conf est très proche de la tienne et ça fonctionne avec mon 3DTouch de triangleLabs. Mais ton modèle est surement différent.
Lors du second palpage, est-ce l'axe Z remonte suffisamment pour que le palpeur puisse se déployer sans toucher le lit? car s'il touche au moment du déploiement, ce qu'il ne doit pas se produire normalement, il se met en erreur et clignote.

Met ça dans ton configuration.h
#define Z_HOMING_HEIGHT  20
#define Z_AFTER_HOMING  20

Et augmente les valeurs suivantes :
#define Z_CLEARANCE_DEPLOY_PROBE   20 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  20 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     20 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           20 // Z position after probing is done

Tu les réduira ensuite pour les optimiser.

Tu as ajouté ça dans ta conf., mais l'interface LCD de MKS en a aussi l'équivalent, en fait le gcode complet pour faire le leveling (par défaut : "G28\nG29\nM500"), je pense qu'il est préférable d'enlever ça pour éviter les confilts :

#define Z_PROBE_END_SCRIPT "M500\nM501"

steph5764, 2021-10-10:
merci de ton aide effectivement  je comptai retirer la variable #define Z_PROBE_END_SCRIPT "M500\nM501" je l'avais trouvé dans une video de mks dans le doute de trouver une solution j'ai essayé . mais là je viens de résoudre le problème en fait il vient de visual code et platformio , après un test avec un autre pc le firmware a fonctionner immédiatement . le plus surprennant c'est que je n avais aucun message d'erreur même en faisant une inspection !!! le principale c'est d'avoir trouvé la solution . 
je te remercie d'avoir pris du temps pour moi cordialement

closed 2021-10-10 by steph5764